maximkulkin/esp-homekit-demo #95: once accessory is removed it can't be re-added

nuxboxen, 2018-06-09:
>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 18:EE:9A:93:4A:C4
>>> HomeKit: Found admin pairing with EC33C480-1E6A-4203-9615-AEC0BA641370, disabling pair setup
>>> HomeKit: Waiting for IP

How do you overcome this situation?

nuxboxen, 2018-06-09:
Sorry for the dumb question,  but if you happen to get stuck in the situation you need to erase.

 make -C examples/led_strip erase_flash

idumzaes, 2018-06-09:
I actually found this great small piece of code to generate MAC addresses. 
I hope it's something you could implement.

```C
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <time.h>

int main() {
  int i, tp;

  srand(time(NULL) + getpid());

  for (i = 0; i < 6; i++) {
    tp = rand() % 256;
    printf("%s%X%s", tp < 16 ? "0" : "", tp, i < 5 ? ":" : "\n");
  }
  return EXIT_SUCCESS;
}
```
source: https://gist.github.com/slim/2401338

maximkulkin, 2018-06-10:
@idumzaes when device knows it was removed from HomeKit (that means that device was online and accessible from Home.app when it was removed), then there is no problem of reconnecting it back, since when Home.app removes accessory, it connects to it and sends command to remove pairing. When last admin pairing is removed, accessory switches back to discovery and pairing mode.

If you removed accessory offline, then you need to factory reset your accessory. Various accessories have different ways to do that. E.g. with Sonoff Basic people usually implement reset on long button press. Some lightbulbs (e.g. LIFX) require you to turn it on and off for 1 second 5 times in a row. So certified hardware also has this reset problem if you do it offline. My suggestion: just don't unpair offline or be prepared to do factory reset/flash erase.

idumzaes, 2018-06-10:
@maximkulkin @RavenSystem
That makes a lot of sense. 
Thanks for the thorough explanation guys.

eqfox, 2018-07-19:
The same issue.

When I added the fourth step of the device, the phone was stuck, and the phone was not added successfully. But the chip starts reporting the same log and the phone cannot be added again.

s it possible to consider not storing the admin pair?

closed 2019-01-27 by maximkulkin

vicfrancco, 2020-08-18:
> homekit_server_reset()

@RavenSystem , could you please give me some hint of how (or where) to use homekit_server_reset() function?

I've tryed to insert it into setup() at ESP32_homekit.ino, but it didn't work.

Thanks.