maximkulkin/esp-homekit-demo #96: Change TYPE - OUTLET - Sonoff Basic

fmoreni, 2018-06-11:
Hello everyone,

Could someone help please?

I am tetando to switch the switch sonoff_basic type to outled, that ICON looks like a socket.

I changed two parameters in the code

  HOMEKIT_ACCESSORY (.id = 1, .category = homekit_accessory_category_switch, .services = (homekit_service_t * []) {
FOR
  HOMEKIT_ACCESSORY (.id = 1, .category = homekit_accessory_category_outled, .services = (homekit_service_t * []) {

  HOMEKIT_SERVICE (SWITCH, .primary = true, .characteristics = (homekit_characteristic_t * []) {
FOR
  HOMEKIT_SERVICE (OULET, .primary = true, .characteristics = (homekit_characteristic_t * []) {

When I look for a new device, I see that it changed the icon to socket, but I can not add it.

What am I doing wrong?

THANK YOU

RavenSystem, 2018-06-11:
You must declare another characteristic, that is mandatory for outlets:
`homekit_characteristic_t switch_outlet_in_use = HOMEKIT_CHARACTERISTIC_(OUTLET_IN_USE, true);
`
And then add it to the service after `&switch_on,`:
`&switch_outlet_in_use,`

fmoreni, 2018-06-11:
It worked perfectly, thank you very much.

RavenSystem, 2018-06-11:
Please, remember to close this issue if your problem has been resolved. ;)

fmoreni, 2018-06-23:
Hello

I'm trying now to change the **Sonoff DUAL**! I used this way
homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, "Dual Lamp");

homekit_characteristic_t switch_outlet_in_use1 = HOMEKIT_CHARACTERISTIC_(OUTLET_IN_USE, true);
homekit_characteristic_t switch_outlet_in_use2 = HOMEKIT_CHARACTERISTIC_(OUTLET_IN_USE, true);

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_outlet, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            &name,
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0"),
            HOMEKIT_CHARACTERISTIC(MODEL, "Dual Lamp"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, lamp_identify),
            NULL
        }),
        HOMEKIT_SERVICE(OUTLET, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Top Light"),
            &top_light_on,
            &switch_outlet_in_use1,
            NULL
        }),
        HOMEKIT_SERVICE(OUTLET, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Bottom Light"),
            &bottom_light_on,
            &switch_outlet_in_use2,
            NULL
        }),
        NULL
    }),
    NULL
};

It did not work! What am I doing wrong?

closed 2018-06-23 by fmoreni

renandw, 2019-10-16:
Can you, please, share your code? Thanks