maximkulkin/esp-homekit-demo #97: Question: Any plans to implement OTA updates?

peros550, 2018-06-13:
Hi, 

I never had the chance, so I would like to thank everybody who with their efforts made this project a reality. 

I was wondering if there are any plans to create an OTA functionality for future updates.

Thanks & Regards

maximkulkin, 2018-06-13:
@HomeACcessoryKid is working on it. I'll leave it to him to explain the progress.

maccoylton, 2018-06-13:
I've been doing some testing and created a few examples based on the work @HomeACcessoryKid has done, a little more work to do on instruction, but I have these up and running across 5 devices.

https://github.com/maccoylton/esp-homekit-sonoff-basic
https://github.com/maccoylton/esp-homekit-sonoff-s20
https://github.com/maccoylton/homekit_wifi_thermostat

closed 2018-08-22 by peros550

kiralikbeyin, 2019-08-03:
@maccoylton do you have any updated repo?

kiralikbeyin, 2019-08-03:
please check #245