maximkulkin/esp-homekit-demo #102: Multiple accessories

sn0wst0rm, 2018-06-25:
Hello, I'm trying to use my esp with your homekit library in order to control two different devices, a fan and a light, both connected to one single esp.
When creating the accessories array, I add a second accessory with `.id=2`, and all its relative services and information.
The firmware compiles fine but when I try to add the two accessories to my home, I am only able to see the first one (with id 1). Moreover it fails to pair, menaing that as soon as the pairing setup is finished, the esp recives an unpair command from homekit and deletes the pairing, resulting in a pair error in home app.

What am I doing wrong?
Thank you!

supersjimmie, 2018-06-25:
Must be something in your code, I have a setup with 4 acc (id 1-4) that works fine.

sn0wst0rm, 2018-06-25:
Could you please provide an example? I cannot really figure out what I am doing wrong. Do you also know if it is possible to dynamically add accessories? Like i connect something to my esp that has the capability to manage other devices (like another light) but then show it in home app only when I actually connet it to my esp

RavenSystem, 2018-06-25:
The best way is to use an unique accessory (.id=1), with one or more services (fan and light are services). And remember to check the required characteristic of each service, documented here: https://github.com/maximkulkin/esp-homekit/blob/master/include/homekit/characteristics.h

maximkulkin, 2018-06-25:
@sn0wst0rm That does not work that way: you can not change list accessories dynamically at runtime (at least, this is not an officially supported behavior).

sn0wst0rm, 2018-06-25:
@RavenSystem yes I've noticed it, using multiple accessory IDs does not work at all, causing accessory pair failure. In addition I can't see how to add a specific password for each of the accessories in the accessories array.

@maximkulkin I meant something like it happens in Homebridge, if one of the plugins discovers a new device, next time you open the home app it shows in the default room and you can use it

peros550, 2018-08-04:
> @maximkulkin I meant something like it happens in Homebridge, if one of the plugins discovers a new device, next time you open the home app it shows in the default room and you can use it

The specific behavior would be very interested to implement in combination with OTA updates.

royfocker19, 2018-08-15:
@supersjimmie could you please share the code were you have 4 different accessories?

peros550, 2018-08-15:
@royfocker19 you may want to check the following: 
https://github.com/peros550/esp-homekit-multiple-sensors

royfocker19, 2018-08-15:
Nice example, I'm doing something wrong. I can compile it. connect to the AP and the to my wireless, but I cannot add the accessories, nothing appears when I want to add them and the led on the ESP doesn't stop blinking.

supersjimmie, 2018-08-16:
@royfocker19 that is pretty straight forward, just repeat things until you have them all.
Use a new ```.id``` for each accessory.
(and don't forget all the ```NULL```s!)

```
homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "MYNAME"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "037A2BABF19D"),
            HOMEKIT_CHARACTERISTIC(MODEL, "supersjimmie"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
            NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary=false, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Room1"),
			&dimbtn3,
			NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Room2"),
			&dimbtn4,
			NULL
        }),
        NULL
    }),

    HOMEKIT_ACCESSORY(.id=2, .category=homekit_accessory_category_thermostat, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Thermostat"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0012346"),
            HOMEKIT_CHARACTERISTIC(MODEL, "Sensors"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
            NULL
        }),
        HOMEKIT_SERVICE(THERMOSTAT, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Airco"),
            &temperature,
//            &humidity,
            &heatstatecurrent,
            &heatstatetarget,
            &temptarget,
            &tempunits,
            NULL
        }),

		NULL
	}),

    HOMEKIT_ACCESSORY(.id=3, .category=homekit_accessory_category_sensor, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Sensors"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0012347"),
            HOMEKIT_CHARACTERISTIC(MODEL, "Sensors"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
            NULL
        }),
        HOMEKIT_SERVICE(HUMIDITY_SENSOR, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Humidity"),
            &humidity,
            NULL
        }),

        HOMEKIT_SERVICE(AIR_QUALITY_SENSOR, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Lucht"),
            &airquality,
			&co2value,
            NULL
        }),
		NULL
	}),

	HOMEKIT_ACCESSORY(.id=4, .category=homekit_accessory_category_sensor, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Light"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "HaPK"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0012349"),
            HOMEKIT_CHARACTERISTIC(MODEL, "LightSensor"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, NULL),
            NULL
        }),
        HOMEKIT_SERVICE(LIGHT_SENSOR, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Licht"),
            &lightvalue,
            NULL
        }),
		NULL
	}),

    NULL
};

```

peros550, 2018-08-16:
@royfocker19 had a similar issue as you described. It turned out for me that my code didn't properly initialize a Light sensor value (withing the default range expected by homekit)  hence I was not able to add it in homekit during the pairing process. I did a lot of troubleshooting to indentify the issue. I was adding one by one my accessories to code, then compile, then try and error to identify what was the root cause of my issue.  What kind of sensors are you trying to integrate?

royfocker19, 2018-08-16:
I’m just adding 2 lightbulbs. Because I want them in different rooms they need to be 2 different accessories. It seems the second accessory is not initializing. I’ll share my code in a few minutes.

royfocker19, 2018-08-16:
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <espressif/esp_common.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <wifi_config.h>

#include "button.h"
#include "toggle.h"

// The GPIO pin that is connected to Triac#1 on the ESP12F V3.
const int relay_gpio_1 = 14;

// The GPIO pin that is connected to Triac#2 on the ESP12F V3.
const int relay_gpio_2 = 16;

// The GPIO pin that is connected to the LED on the ESP12F V3.
const int led_gpio = 2;

// The GPIO pin that is connected to the button on the ESP12F V3.
const int button_gpio = 0;

// The GPIO pin that is connected to the header on the ESP12F V3 (external switches).
const int toggle_gpio_1 = 12;
const int toggle_gpio_2 = 13;

homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, "ESP12F V3");

void light_identify_task(void *_args) {
    // We identify the ESP12F by flashing it's LED.
    for (int i=0; i<3; i++) {
        for (int j=0; j<2; j++) {
            led_write(true);
            vTaskDelay(200 / portTICK_PERIOD_MS);
            led_write(false);
            vTaskDelay(200 / portTICK_PERIOD_MS);
        }
        vTaskDelay(500 / portTICK_PERIOD_MS);
    }
    led_write(false);
    vTaskDelete(NULL);
}

void light_identify(homekit_value_t _value) {
    printf("Light identify\n");
    xTaskCreate(light_identify_task, "Light identify", 128, NULL, 2, NULL);
}

void button_callback(uint8_t gpio, button_event_t event);

void led_write(bool on) {
    gpio_write(led_gpio, on ? 0 : 1);
}

void reset_configuration_task() {
    //Flash LED first before we start the reset
    for (int i=0; i<3; i++) {
        led_write(true);
        vTaskDelay(100 / portTICK_PERIOD_MS);
        led_write(false);
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }
    printf("Resetting Wifi   due to button long   at GPIO %2d\n", button_gpio);
    wifi_config_reset();
[... 171 lines cut ...]
    gpio_enable(button_gpio,   GPIO_INPUT);
    gpio_enable(toggle_gpio_1, GPIO_INPUT);
    gpio_enable(toggle_gpio_2, GPIO_INPUT);
}

void user_init(void) {
    uart_set_baud(0, 115200);
    wifi_config_init("ESP12F V3", NULL, on_wifi_ready);
    gpio_init();
    create_accessory_name();
    

// 4000 = 4s long press => 4000/4 = 1s medium press
    if (button_create(button_gpio, 0, 4000, button_callback)) {
        printf("Failed to initialize button\n");
    }
    
    if (toggle_create(toggle_gpio_1, toggle_callback_1)) {
        printf("Failed to initialize toggle 1 \n");
    }

    if (toggle_create(toggle_gpio_2, toggle_callback_2)) {
        printf("Failed to initialize toggle 2 \n");
    }
}

royfocker19, 2018-08-16:
Am I missing some Null or something?

royfocker19, 2018-08-16:
I think there were 3 NULLs missing....

supersjimmie, 2018-08-17:
yep,  just like I suggested: "don't forget all the NULLs!".  👍

royfocker19, 2018-08-17:
Something that I don't understand is why the accessories appear in a
different order from the one in the code.
Regards
Fede

closed 2019-01-27 by maximkulkin

Yousefff1, 2019-04-29:
> Ok, it worked, final code is this.
> 
> #include <stdio.h>
> #include <espressif/esp_wifi.h>
> #include <espressif/esp_sta.h>
> #include <espressif/esp_common.h>
> #include <esp/uart.h>
> #include <esp8266.h>
> #include <FreeRTOS.h>
> #include <task.h>
> 
> #include <homekit/homekit.h>
> #include <homekit/characteristics.h>
> #include <wifi_config.h>
> 
> #include "button.h"
> #include "toggle.h"
> 
> // The GPIO pin that is connected to triac#1 on the ESP12F V3.
> const int triac_gpio_1 = 14;
> 
> // The GPIO pin that is connected to triac#2 on the ESP12F V3.
> const int triac_gpio_2 = 16;
> 
> // The GPIO pin that is connected to the LED on the ESP12F V3.
> const int led_gpio = 2;
> 
> // The GPIO pin that is connected to the button on the ESP12F V3.
> const int button_gpio = 0;
> 
> // The GPIO pin that is connected to the header on the ESP12F V3 (external switch).
> const int toggle_gpio_1 = 12;
> const int toggle_gpio_2 = 13;
> 
> homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, "ESP12F V3");
> 
> void led_write(bool on) {
> gpio_write(led_gpio, on ? 0 : 1);
> }
> 
> void light_identify_task(void *_args) {
> // We identify the ESP12F by Flashing it's LED.
> for (int i=0; i<3; i++) {
> for (int j=0; j<2; j++) {
> led_write(true);
> vTaskDelay(200 / portTICK_PERIOD_MS);
> led_write(false);
> vTaskDelay(200 / portTICK_PERIOD_MS);
> }
> vTaskDelay(500 / portTICK_PERIOD_MS);
> }
> led_write(false);
> vTaskDelete(NULL);
> }
> 
> void light_identify(homekit_value_t _value) {
> printf("Light identify\n");
> xTaskCreate(light_identify_task, "Light identify", 128, NULL, 2, NULL);
> }
> 
> void button_callback(uint8_t gpio, button_event_t event);
> 
> void reset_configuration_task() {
> //Flash LED first before we start the reset
> for (int i=0; i<3; i++) {
> led_write(true);
> vTaskDelay(100 / portTICK_PERIOD_MS);
> led_write(false);
> vTaskDelay(100 / portTICK_PERIOD_MS);
> }
[... 207 lines cut ...]
> wifi_config_init("ESP12F V3", NULL, on_wifi_ready);
> gpio_init();
> create_accessory_name();
> 
> // 4000 = 4s long press => 4000/4 = 1s medium press
> if (button_create(button_gpio, 0, 4000, button_callback)) {
> printf("Failed to initialize button\n");
> }
> 
> ```
> if (toggle_create(toggle_gpio_1, toggle_callback_1)) {
>     printf("Failed to initialize toggle 1 \n");
> }
> 
> if (toggle_create(toggle_gpio_2, toggle_callback_2)) {
>     printf("Failed to initialize toggle 2 \n");
> }
> ```
> }

Hello,

Thank you for sharing this, I keep getting errors trying to apply this code, would you mind sharing the whole directory files please ?

also can we increase the number of relays controlled by this to say 10 ?

okanduzyel, 2019-10-14:
I would like to use 5 different sonoff basic application in my home. 

5 different sonoff, 5 different esp8266. Should I change anything on code? If I program all of them with same sonoff_basic example code, Do they conflict?

maximkulkin, 2019-10-14:
@okanduzyel  You should be fine with flashing same firmware to all of them

spxnr, 2020-05-18:
I am flashing the same image to a couple esp8266's and only one is showing up on Home app.. But when I change the Name characteristic to something unique, the second one shows up. I tried just incrementing the Serial Number, but that didn't solve the problem like changing the Name did. Any ideas on why?

royfocker19, 2020-05-18:
Could you please share the code? It’s a multiple accessory like the one discussed here?

maximkulkin, 2020-05-18:
@spxnr by writing here I think you're confusing what "multiple accessories" meant by original issue author. He meant the case when you configure esp-homekit with an array of more than one accessory, so single device acts as many accessories. Your problem as far as I understand is making multiple devices co-exist in the same network. I do not know which firmware you're trying to use (my guess is your own one), so I suggest checking something like sonoff_basic firmware [handles name uniqueness](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/sonoff_basic/main.c#L176-L187).

ttaskin, 2020-07-20:
@maximkulkin hi, i just start to use esp32 and i use your code about esp32 homekit (builtin_led_accessory) , i confused about to use multiple switch, do you have any suggestion, or do you have any example about it, i just want to control 8 relay with esp32 and homekit, i try to understand royfocker19's code but i fail. so many library doesent exist in my arduino ide. and the code for esp8266.

royfocker19, 2020-07-21:
Share your code so I can truly to check it

ttaskin, 2020-07-21:
@royfocker19  im really level 0 person about esp and my code is ;

[2 pin control with esp.txt](https://github.com/maximkulkin/esp-homekit-demo/files/4955213/2.pin.control.with.esp.txt)
/*
 * builtin_led_accessory.c
 * Define the accessory in pure C language using the Macro in characteristics.h
 *
 *  Created on: 2020-04-13
 *      Author: Mixiaoxiao (Wang Bin)
 */

#include <Arduino.h>
#include <homekit/types.h>
#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <stdio.h>
#include <port.h>
#include <esp_wifi_types.h>
#include <esp_wifi.h>

static char ACCESSORY_NAME[33] = "TT AKILLI EV SİSTEMİ 2";
#define ACCESSORY_SN  ("00002")  //SERIAL_NUMBER
#define ACCESSORY_MANUFACTURER ("Tarkan TAŞKIN")
#define ACCESSORY_MODEL  ("LED 2")

#define PIN_LED2 23

static char ACCESSORY_NAME[32] = "TT AKILLI EV SİSTEMİ";
#define ACCESSORY_SN  ("00001")  //SERIAL_NUMBER
#define ACCESSORY_MANUFACTURER ("Tarkan TAŞKIN")
#define ACCESSORY_MODEL  ("LED")

#define PIN_LED 2

void led_set_power(bool on) {
	digitalWrite(PIN_LED, on ? HIGH : LOW);
	digitalWrite(PIN_LED2, on ? HIGH : LOW);
}

void led_setter(homekit_value_t value);

homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, ACCESSORY_NAME);
homekit_characteristic_t serial_number = HOMEKIT_CHARACTERISTIC_(SERIAL_NUMBER, ACCESSORY_SN);
homekit_characteristic_t led_on = HOMEKIT_CHARACTERISTIC_(ON, false, .setter=led_setter);

void led_setter(homekit_value_t value) {
	if (value.format != homekit_format_bool) {
		printf("Invalid on-value format: %d\n", value.format);
		return;
	}
	const bool power = value.bool_value;
	led_on.value.bool_value = power; // Sync the value, and no getter needed.
	led_set_power(power);
	printf("Set Builtin-LED to %s\n", power ? "ON" : "OFF");
}

void led_toggle() {
	bool power = !led_on.value.bool_value;
	led_on.value.bool_value = power;
	led_set_power(power);
	homekit_characteristic_notify(&led_on, led_on.value);
}

void led_blink_task(void *_args) {
	for (int i = 0; i < 3; i++) {
		led_set_power(true);
		vTaskDelay(100 / portTICK_PERIOD_MS);
		led_set_power(false);
		vTaskDelay(100 / portTICK_PERIOD_MS);
	}
	led_set_power(false);
	vTaskDelete(NULL);
}

void led_blink() {
	xTaskCreate(led_blink_task, "led_blink_task", 2048, NULL, 1, NULL);
}

void accessory_identify(homekit_value_t _value) {
	printf("accessory identify\n");
	led_blink();
}

homekit_accessory_t *accessories[] = {
				HOMEKIT_ACCESSORY(
						.id = 1,
						.category = homekit_accessory_category_lightbulb,
						.services=(homekit_service_t*[]){
						  HOMEKIT_SERVICE(ACCESSORY_INFORMATION,
						  .characteristics=(homekit_characteristic_t*[]){
						    &name,
						    HOMEKIT_CHARACTERISTIC(MANUFACTURER, ACCESSORY_MANUFACTURER),
						    &serial_number,
						    HOMEKIT_CHARACTERISTIC(MODEL, ACCESSORY_MODEL),
						    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0.1"),
						    HOMEKIT_CHARACTERISTIC(IDENTIFY, accessory_identify),
						    NULL
						  }),
						  HOMEKIT_SERVICE(LIGHTBULB, .primary=true,
						  .characteristics=(homekit_characteristic_t*[]){
						    HOMEKIT_CHARACTERISTIC(NAME, "Led"),
						    &led_on,
						    NULL
						  }),
						  NULL
						}),
				NULL
HOMEKIT_ACCESSORY(
						.id = 2,
						.category = homekit_accessory_category_lightbulb,
						.services=(homekit_service_t*[]){
						  HOMEKIT_SERVICE(ACCESSORY_INFORMATION,
						  .characteristics=(homekit_characteristic_t*[]){
						    &name,
						    HOMEKIT_CHARACTERISTIC(MANUFACTURER, ACCESSORY_MANUFACTURER),
						    &serial_number,
						    HOMEKIT_CHARACTERISTIC(MODEL, ACCESSORY_MODEL),
						    HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0.1"),
						    HOMEKIT_CHARACTERISTIC(IDENTIFY, accessory_identify),
						    NULL
						  }),
						  HOMEKIT_SERVICE(LIGHTBULB, .primary=true,
						  .characteristics=(homekit_characteristic_t*[]){
						    HOMEKIT_CHARACTERISTIC(NAME, "Led2"),
						    &led_on,
						    NULL
						  }),
						  NULL
						}),
				NULL
		};

homekit_server_config_t config = {
		.accessories = accessories,
		.password = "813-55-297",
		//.on_event = on_homekit_event,
		.setupId = "ABCD"
};

void accessory_init() {
	pinMode(PIN_LED, OUTPUT);
	pinMode(PIN_LED2, OUTPUT);
	led_set_power(false);
	//Rename ACCESSORY_NAME base on MAC address.
	uint8_t mac[6];
	esp_wifi_get_mac(WIFI_IF_STA, mac);
	sprintf(ACCESSORY_NAME, "ESP32_LED_%02X%02X%02X", mac[3], mac[4], mac[5]);
}

maximkulkin, 2020-07-21:
I think I can see a problem here:

[image]

ttaskin, 2020-07-21:
how can i fix it ? do you have any suggestion ?

maximkulkin, 2020-07-21:
Do proper formatting and the problem will become obvious.

ttaskin, 2020-07-21:
thank you @maximkulkin  first step ok now i can see 2 device on iphone screen but one of them say doesn't respons, and other switch control both pin. how can i seperate it ?

#include <Arduino.h>
#include <homekit/types.h>
#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <stdio.h>
#include <port.h>
#include <esp_wifi_types.h>
#include <esp_wifi.h>

static char ACCESSORY_NAME[32] = "TT AKILLI EV SİSTEMİ";
#define ACCESSORY_SN  ("00001")  //SERIAL_NUMBER
#define ACCESSORY_MANUFACTURER ("Tarkan TAŞKIN")
#define ACCESSORY_MODEL  ("LED")

#define PIN_LED 2

#define ACCESSORY_MODEL  ("LED2")
#define PIN_LEDd 23
void led_set_power(bool on) {
  digitalWrite(PIN_LED, on ? HIGH : LOW);

  digitalWrite(PIN_LEDd, on ? HIGH : LOW);
}
void led_setter(homekit_value_t value);

homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, ACCESSORY_NAME);
homekit_characteristic_t serial_number = HOMEKIT_CHARACTERISTIC_(SERIAL_NUMBER, ACCESSORY_SN);
homekit_characteristic_t led_on = HOMEKIT_CHARACTERISTIC_(ON, false, .setter=led_setter);

void led_setter(homekit_value_t value) {
  if (value.format != homekit_format_bool) {
    printf("Invalid on-value format: %d\n", value.format);
    return;
  }
  const bool power = value.bool_value;
  led_on.value.bool_value = power; // Sync the value, and no getter needed.
  led_set_power(power);
  printf("Set Builtin-LED to %s\n", power ? "ON" : "OFF");
}

void led_toggle() {
  bool power = !led_on.value.bool_value;
  led_on.value.bool_value = power;
  led_set_power(power);
  homekit_characteristic_notify(&led_on, led_on.value);
}

void led_blink_task(void *_args) {
  for (int i = 0; i < 3; i++) {
    led_set_power(true);
    vTaskDelay(100 / portTICK_PERIOD_MS);
    led_set_power(false);
    vTaskDelay(100 / portTICK_PERIOD_MS);
  }
  led_set_power(false);
  vTaskDelete(NULL);
}

void led_blink() {
  xTaskCreate(led_blink_task, "led_blink_task", 2048, NULL, 1, NULL);
}

void accessory_identify(homekit_value_t _value) {
  printf("accessory identify\n");
  led_blink();
}

homekit_accessory_t *accessories[] = {
        HOMEKIT_ACCESSORY(
            .id = 1,
            .category = homekit_accessory_category_lightbulb,
            .services=(homekit_service_t*[]){
              HOMEKIT_SERVICE(ACCESSORY_INFORMATION,
              .characteristics=(homekit_characteristic_t*[]){
                &name,
                HOMEKIT_CHARACTERISTIC(MANUFACTURER, ACCESSORY_MANUFACTURER),
                &serial_number,
                HOMEKIT_CHARACTERISTIC(MODEL, ACCESSORY_MODEL),
                HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0.1"),
                HOMEKIT_CHARACTERISTIC(IDENTIFY, accessory_identify),
                NULL
              }),
              HOMEKIT_SERVICE(LIGHTBULB, .primary=true,
              .characteristics=(homekit_characteristic_t*[]){
                HOMEKIT_CHARACTERISTIC(NAME, "Led"),
                &led_on,
                NULL
              }),
              NULL
            }),
        HOMEKIT_ACCESSORY(
            .id = 2,
            .category = homekit_accessory_category_lightbulb,
            .services=(homekit_service_t*[]){
              HOMEKIT_SERVICE(ACCESSORY_INFORMATION,
              .characteristics=(homekit_characteristic_t*[]){
                &name,
                HOMEKIT_CHARACTERISTIC(MANUFACTURER, ACCESSORY_MANUFACTURER),
                &serial_number,
                HOMEKIT_CHARACTERISTIC(MODEL, ACCESSORY_MODEL),
                HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0.1"),
                HOMEKIT_CHARACTERISTIC(IDENTIFY, accessory_identify),
                NULL
              }),
              HOMEKIT_SERVICE(LIGHTBULB, .primary=true,
              .characteristics=(homekit_characteristic_t*[]){
                HOMEKIT_CHARACTERISTIC(NAME, "Led"),
                &led_on,
                NULL
              }),
              NULL
            }),
        NULL
    
    };

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "813-55-297",
    //.on_event = on_homekit_event,
    .setupId = "ABCD"
};

void accessory_init() {
  pinMode(PIN_LED, OUTPUT);
   pinMode(PIN_LEDd, OUTPUT);

ttaskin, 2020-07-22:
ok i did it finaly :)) and here it is .
1st link updated 2 component
2nd link original project
[Arduino-HomeKit-ESP32-2 component updated.zip](https://github.com/maximkulkin/esp-homekit-demo/files/4957185/Arduino-HomeKit-ESP32-2.component.updated.zip)
[Arduino-HomeKit-ESP32-master.zip](https://github.com/maximkulkin/esp-homekit-demo/files/4957186/Arduino-HomeKit-ESP32-master.zip)

royfocker19, 2020-07-22:
Hi, qq, for Led2 you have &led_on and for Led you have &led2_on,  is that right?

Press in Dire Situations http://www.nooooooooooooooo.com/

ttaskin, 2020-07-22:
@royfocker19  really ? i don't see like thet situation, by the way im realllllyy level 0 at programing and esp :)) 
may you show my wrong at my code please.

ttaskin, 2020-07-22:
ok ok i find it :) at the end of code, thank you for your attension

AramVartanyan, 2020-07-22:
Verify your version with the following example:

‘’’
homekit_characteristic_t switch1_on = HOMEKIT_CHARACTERISTIC_(ON, false, .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(switch1_on_callback));

homekit_characteristic_t switch2_on = HOMEKIT_CHARACTERISTIC_(ON, false, .callback=HOMEKIT_CHARACTERISTIC_CALLBACK(switch2_on_callback));

homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, D_MODEL);

homekit_characteristic_t name1 = HOMEKIT_CHARACTERISTIC_(NAME, D_NAME1);

homekit_characteristic_t name2 = HOMEKIT_CHARACTERISTIC_(NAME, D_NAME2);

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            &name,
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, D_MANIFACTURER),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, D_SN),
            HOMEKIT_CHARACTERISTIC(MODEL, D_MODEL),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, D_FW),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, switch_identify),
            NULL
        }),
                HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, D_NAME1),
            &name1,
            &switch1_on,
            NULL
        }),
        NULL
    }),
    HOMEKIT_ACCESSORY(.id=2, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            &name,
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, D_MANIFACTURER),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, D_SN),
            HOMEKIT_CHARACTERISTIC(MODEL, D_MODEL),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, D_FW),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, switch_identify),
            NULL
        }),
            HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, D_NAME2),
            &name2,
            &switch2_on,
            NULL
        }),
        NULL
    }),
    NULL
};
‘’’

paladin32776, 2020-12-07:
> Hello, I'm trying to use my esp with your homekit library in order to control two different devices, a fan and a light, both connected to one single esp.
> When creating the accessories array, I add a [...]

Looks like there cannot be two HomeKit accessories in a home with the same ACCESSORY_INFORMATION service names.

If you have it that name hard coded you probably have to make it dynamic, unless you wanna change the source code for every piece of hardware added. 

I added a client_name field to my captive page, put the value in a string, and point the corresponding homekit_characteristic_t to it, like so:

    homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_(NAME, "initial name");

later in my code, after the Wifi connection has been established, but before HomeKit initialization:

    cha_name.value.string_value = client_name;

Maybe that helps.

askarkurymbayev, 2021-12-22:
> Looks like there cannot be two HomeKit accessories in a home with the same ACCESSORY_INFORMATION service names.
> 
> If you have it that name hard coded you probably have to make it dynamic, unless you wanna change the source code for every piece of hardware added.
> 
> I added a client_name field to my captive page, put the value in a string, and point the corresponding homekit_characteristic_t to it, like so:
> 
> ```
> homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_(NAME, "initial name");
> ```
> 
> and
> 
> ```
> homekit_accessory_t *accessories[] = {
>     HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=.   (homekit_service_t*[])
> 		{
>         HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[])
> 				{
>             &cha_name,
> ....
> ```
> 
> later in my code, after the Wifi connection has been established, but before HomeKit initialization:
> 
> ```
> cha_name.value.string_value = client_name;
> ```
> 
> Maybe that helps.

Good day!
could you please share the code, I can't make a dynamic name for the devices