maximkulkin/esp-homekit-demo #103: How to send station ip?

kiralikbeyin, 2018-06-27:
Is where anyway to show station ip of esp somewhere in HomeKit app?

Also please can you give an example of LOGS? --- HOMEKIT_DECLARE_CHARACTERISTIC_LOGS

maximkulkin, 2018-06-28:
> Is where anyway to show station ip of esp somewhere in HomeKit app?

Why would you want this? Accessory IP can be easily determined from mDNS. You are not expected to connect to accessory with an internet browser.

Second, I do not have any example on LOGS characteristic. And it is not documented in official documentation. And unless you build a Lock management service or building your own iOS application to interface with HomeKit, there is no point in using LOGS characteristic.

kiralikbeyin, 2018-06-28:
My project uses a static ip and sonoff gets http requests. 
If there are 10 sonoffs so i can find ip addr easly by name with HomeKit app

I found this but dont know how to send to HomeKit

```
static const char* get_my_ip(void)
{
    static char ip[16] = "0.0.0.0";
    ip[0] = 0;
    struct ip_info ipinfo;
    (void) sdk_wifi_get_ip_info(STATION_IF, &ipinfo);
    snprintf(ip, sizeof(ip), IPSTR, IP2STR(&ipinfo.ip));
    return  (char*) ip;
}
```

I will try Lock management but if it shows like a lock in HomeKit app, will not be usefull.

There is code about AUDIO , can esp send audio?

NorthernMan54, 2018-06-28:
Over on the homebridge/hap-nodejs side, I have been using the serial number characteristic to show the IP address of the accessory.  It shows up in the Home app.

kiralikbeyin, 2018-07-03:
I got  error: initializer element is not constant, how can i pass ip to MANUFACTURER? I dont know how to convert..

```

static const char* get_my_ip(void)
{
    static char ip[16] = "0.0.0.0";
    ip[0] = 0;
    struct ip_info ipinfo;
    (void) sdk_wifi_get_ip_info(STATION_IF, &ipinfo);
    snprintf(ip, sizeof(ip), IPSTR, IP2STR(&ipinfo.ip));
    return  (char*) ip;
}

homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, "EvAk Switch");
homekit_characteristic_t serial = HOMEKIT_CHARACTERISTIC_(SERIAL_NUMBER, "EvAk N/A");
homekit_characteristic_t ne = HOMEKIT_CHARACTERISTIC_(MANUFACTURER, get_my_ip());

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            &name,
          //  HOMEKIT_CHARACTERISTIC(MANUFACTURER, get_my_ip() ),
            &ne,
            &serial,
            HOMEKIT_CHARACTERISTIC(MODEL, "Basic"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, identify),
            NULL
        }),
        HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Switch"),
            &switch_on,
            NULL
        }),
        NULL
    }),
    NULL
};
```

/esp-homekit-devices/devices/1/main.c:171:1: error: initializer element is not constant
 homekit_characteristic_t ne = HOMEKIT_CHARACTERISTIC_(MANUFACTURER, get_my_ip());
 ^
/esp-homekit-devices/devices/1/main.c:171:1: error: (near initialization for 'ne.value.<anonymous>.string_value')
make: *** [build/program//main.o] Error 1

hn-nguyen, 2018-07-03:
I have not tested the following code, but I think it can give you the idea:

```C
homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, "Switch");
homekit_characteristic_t serial = HOMEKIT_CHARACTERISTIC_(SERIAL_NUMBER, "N/A");
homekit_characteristic_t ne = HOMEKIT_CHARACTERISTIC_(MANUFACTURER,  NULL);  //no IP at the first time

static void get_my_ip(void){
    static char ip[16];
    ip[0] = 0;
    struct ip_info ipinfo;
    (void) sdk_wifi_get_ip_info(STATION_IF, &ipinfo);
    snprintf(ip, sizeof(ip), IPSTR, IP2STR(&ipinfo.ip));
    //copy IP to ne value
    ne.value.string_value = strdup( ip );
}

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            &name,
            &ne,
            &serial,
            HOMEKIT_CHARACTERISTIC(MODEL, "Basic"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, identify),
            NULL
        }),
        HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Switch"),
            &switch_on,
            NULL
        }),
        NULL
    }),
    NULL
};

void on_wifi_ready(){
  get_my_ip();
  ...
  homekit_server_init( .. );
}
```

kiralikbeyin, 2018-07-03:
It worked. Thank you!

closed 2018-07-03 by kiralikbeyin