maximkulkin/esp-homekit-demo #108: Тhermostat does not respond ?

Kristian8606, 2018-07-24:
A few minutes after adding a thermostat does not respond in the home application,
but the monitor shows that it works and even receives temperature data.
Has anyone had the same problem and how to fix it?

ruudk, 2018-07-24:
I have the same issue, it works great for a couple of times but after +-30 it stops working. I am running it on a esp32, you?

sorriso93, 2018-07-24:
I’m running it on wemos d1 mini...

Kristian8606, 2018-07-24:
NodeMCU

maximkulkin, 2018-07-28:
Hey @maccoylton, that issue with updates on integer characteristics should be fixed in latest esp-homekit.

maccoylton, 2018-07-29:
@max thanks latest version does seem to fix that issue

peros550, 2018-07-31:
Hi, I am sorry for this basic question. How do I make sure I am using the latest version?

ruudk, 2018-07-31:
Run ```git submodule update --init --recursive```

Kristian8606, 2018-07-31:
Hello  I will ask my question here.
My wife does not have an ios device and can not control a thermostat.
I would like to add two buttons to raise and lower the temperature.
Is this the variable I need to increase and decrease -> target_temperature.value.float_value?

Please give your conversion suggestions.

ruudk, 2018-07-31:
Yes, and then call ` homekit_characteristic_notify`

See my example repository (has a temperature sensor, but it's the same behavior)

https://github.com/ruudk/esp32-homekit-contact-and-temperature-sensor/blob/master/main/main.c#L263:L264

Kristian8606, 2018-07-31:
I tried your advice -> git submodule update --init --recursive but the device still does not respond

Kristian8606, 2018-08-04:
I had to reinstall Mac OS
I installed everything blinking again on ESP, but the thermostat still does not respond.
What should I do ?

ruudk, 2018-08-05:
Did you remove the device and then add it again in the Home app?

Kristian8606, 2018-08-05:
This thermostat works for me (https://github.com/RavenSystem/esp-homekit-devices/tree/master/devices_legacy/Sonoff_TH)

maximkulkin, 2018-08-06:
@Kristian8606 you’re doing something wrong. Just reflashed my thermostat with latest code version and it is running no problem for a day so far

Kristian8606, 2018-08-06:
Should I use the latest version after I install everything again?
I do everything according to the instructions.

closed 2019-02-26 by maximkulkin