maximkulkin/esp-homekit-demo #112: Question: Motion Sensor Accessory

peros550, 2018-08-01:
I'm thinking to build a motion sensor accessory.  
The sensor would be something like [this HC-SR501](https://learn.adafruit.com/pir-passive-infrared-proximity-motion-sensor?view=all) and if bought on ebay cost would be around 1USD.

As my esp platform I would use Wemos D1 mini.  

There is no special library needed for the sensor to work. It basically has one digital output which turns high  (or low depending on the selected logic) when motion is detected. 

My question would be, how could I create a homekit "motion" accessory in the first place.

maccoylton, 2018-08-01:
See my example here:- https://github.com/maccoylton/esp-homekit-motion-sensor

peros550, 2018-08-02:
Thanks it worked perfectly! I'm going to study your implementation and see if I can enhance it in any way. 
My goal would be to have a few accessories (like temperature, motion and light) integrated in the same hardware device. 

@renssies Many thanks for the tip. I will try AM312, just need to wait a few weeks until I receive it :)

maccoylton, 2018-08-02:
You can fix the interference issue on the HC-SR501 with a solution I found with an internet search 

"You simple need to solder a 220nF ceramic capacitor across pin12 und pin13 of the chip (BISS0001) so that the high-frequency interference coming from the ESP8266 don't influece the gain stage anymor".

peros550, 2018-08-04:
@maccoylton here is a multi sensor example I have just created. 

https://github.com/peros550/esp-homekit-multiple-sensors

Many thanks for your help. I hope this example could be helpful to anyone.

closed 2018-08-22 by peros550