maximkulkin/esp-homekit-demo #113: error building led_strip on ESP-01

dezdeepblue, 2018-08-02:
I just follow the routine and build the led_strip on ESP-01. The flash size is 1MB. Checked with two ESP-01 and both failed to pair. flash ids devices are 3014 and 4014.
I've made the changes as states in the tutorial:
uncommenting FLASH_SIZE ?= 8 and HOMEKIT_SPI_FLASH_BASE_ADDR ?= 0x7A000 and commenting FLASH_SIZE ?= 32.
But in the pairing process this happens and the pairing failed:
dhcp client start...
ip:192.168.1.33,mask:255.255.255.0,gw:192.168.1.1
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
!!! HomeKit: Failed to compact data: sector data read error
!!! HomeKit: Failed to write pairing info to flash: max number of pairings
!!! HomeKit: [Client 4] Failed to store pairing (code -2)
>>> HomeKit: [Client 4] Closing client connection

I then build it with proper configuration on Wemos D1 mini and it worked perfectly.
Do I miss something or the ESP-01 isn't just supported

konifer44, 2018-08-07:
Try to erase flash before flashing.

dezdeepblue, 2018-08-09:
I erased the flash before programming the new firmware but didn't help.
It is now couldn't even boot properly!
Does this compilable/compatible with 1MB flash sizes? Has anyone successfully tested this?
I'm sorry @maximkulkin , have you test it? can you help?

maximkulkin, 2018-08-09:
Ok, I still see that it tries to use flash address 0x100000:
> !!! HomeKit: Failed to read flash magic
> HomeKit: Formatting flash at 0x100000

Make sure you do clean build (`make -C examples/led_strip clean all`) after setting those environment variables.

dezdeepblue, 2018-08-09:
Thank you for your great support

dezdeepblue, 2018-08-09:
>

closed 2018-08-09 by dezdeepblue

sultan95678, 2019-08-23:
hi
thank you for you'r helps.
i have a problem in my project.
i have ESP8266 node MCU
and i want to connected it with homekit
i follow this tutorial : https://www.youtube.com/watch?v=QBj8OLig8Kg&list=WL&index=44&t=0s
every think is good to me , but when i arrive in this steep : 
( vagrant@vagrant-ubuntu-trusty-64:~/esp-homekit-demo$ make -C examples/led tast )
its show me this ( serial.serialutil.SerialException: could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'
make: *** [flash] Error 1
make: Leaving directory `/home/vagrant/esp-homekit-demo/examples/led' )
can you help me in this ERROR

konifer44, 2019-08-23:
What system do You use? Windows/Linux/MacOS

konifer44, 2019-08-23:
You have selected wrong serial port 😉

sultan95678, 2019-08-23:
hi ,, windows

sultan95678, 2019-08-23:
how can i fix it

Alr2303, 2019-08-24:
Did the esp chip shows up in virtualbox, if yes then try reconnecting the usb if no try installing ch34 driver else try installing the whole method once again the video has less info for your issues

sultan95678, 2019-08-24:
thank you
but still the same problem
i have install ch34 driver , and i have restart my computer.
and i have start from the first step
this is the file i have ( vagrant up )
maybe you can edit for me
https://drive.google.com/file/d/1oX8YuvKLJpnNvIhJibN4VGN90wiPShBK/view

Alr2303, 2019-08-24:
u really dont need to edit the code 
its the port where the NodeMCU is connected
look at the video at "3:16" to "2:28"
send the screenshots of the "virtualbox"
like these 
https://imgbbb.com/image/F5knl
https://imgbbb.com/image/F5JTn

sultan95678, 2019-08-24:
take : https://imgbbb.com/image/FEtpG

sultan95678, 2019-08-24:
and this CMD is look like: https://imgbbb.com/image/FE1Sc

maximkulkin, 2019-08-24:
@sultan95678 you miss the note in original repository that you USB port
might be different and you need to adjust. What I usually do is first do
`ls /dev/tty*` before connecting USB and then same command - after
connecting USB and find out what new device appears- this is your port. If
device does not appear at all, you need to check if you have proper drivers
both on host system and Linux inside virtual machine.

sultan95678, 2019-08-24:
it is show from (this pc > device manager then (universal serial bus controllers) 
when i connecting the ESP8266 there is show a new device named ( VirtualBox USB )

sultan95678, 2019-08-25:
@maximkulkin @Alr2303 
I don't know where is the problem.
can any one tell me.

maximkulkin, 2019-08-25:
Sorry, I do not have or use Windows

sultan95678, 2019-08-25:
@maximkulkin  thanks for you'r help

Alr2303, 2019-08-26:
@sultan95678 could u send me the cable nodemcu pics with your windows version

Alr2303, 2019-08-26:
Are u new to linux if yes first learn abt linux and its commands , to understand this code and working u need to understand linux commands, are u using the virtualbox version 5.x or the newer one, if u could send me any video of this whole, i could help u out

gazodia, 2020-06-10:
Hey there,
I'm currently trying to run the basic led example in my esp8266 (esp01). I'm getting the same error as @dezdeepblue described in the first issue:

Opening /dev/tty.usbserial-1420 at 115200bps...
ip:192.168.0.27,mask:255.255.255.0,gw:192.168.0.1
>>> HomeKit: Got new client connection: 4 from 192.168.0.10
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
!!! HomeKit: Failed to compact HomeKit storage: sector data read error
!!! HomeKit: Failed to write pairing info to HomeKit storage: max number of pairings
!!! HomeKit: [Client 4] Failed to store pairing (code -2)
>>> HomeKit: [Client 4] Closing client connection

I already tried the solutions above, but none of them worked for me. No matter what I do the same error still coming up. Do you guys have any idea what it may be?

maximkulkin, 2020-06-11:
What does it say when it starts up? Which address it uses?

gazodia, 2020-06-11:
Currently I'm using the address "0xF5000" that you recommended. No issues while erasing, building or flashing, the problem appears when I try to pair the esp to the Homekit app in my phone. That's exactly what I get when I monitor the connection:

@MacBook-Pro-de-Gabriel esp-homekit-demo % make -C examples/led monitor
/Users/gabrielgazal/esp/esp-open-rtos//utils/filteroutput.py --port /dev/tty.usbserial-1420 --baud 115200 --elf ./build/led.out
Opening /dev/tty.usbserial-1420 at 115200bps...
>>> HomeKit: Got new client connection: 4 from 192.168.0.10
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
!!! HomeKit: Failed to compact HomeKit storage: sector data read error
!!! HomeKit: Failed to write pairing info to HomeKit storage: max number of pairings
!!! HomeKit: [Client 4] Failed to store pairing (code -2)
>>> HomeKit: [Client 4] Closing client connection

maximkulkin, 2020-06-11:
Do not worry about deprecated flash size arguments, it still works.

I would suggest capturing log as soon as device starts (as you miss an important messages). I usually issue both "flash" and "monitor" targets at the same time so it starts monitoring as soon as flashing is finished:
```sh
# set SDK_PATH environment variable
make -C examples/led clean
docker-run esp-rtos make -C examples/led all
make -C examples/led erase_flash
make -C examples/led flash monitor
```

(docker-run is [my helper function](https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP8266-(Docker)) to simplify running stuff in Docker containers)

The other way to get all the messages is to launch monitor and then press and release reset button on ESP module (if your module has one):
[image]

```
ESP-Open-SDK ver: 0.0.1 compiled @ Jun 11 2020 05:02:35
phy ver: 273, pp ver: 8.3

mode : sta(60:01:94:31:68:f0)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Formatting HomeKit storage at 0x100000
>>> HomeKit: Generated new accessory ID: 3F:F6:38:C8:D1:42
scandone
add 0
aid 2
cnt

connected with MyWiFi, channel 1
dhcp client start...
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Sample LEDmd=MyLEDpv=1.0id=3F:F6:38:C8:D1:42c#=1s#=1ff=0sf=1ci=5 Port=5556 TTL=4500
ip:10.0.1.123,mask:255.255.255.0,gw:10.0.1.1
```

Notice this line:
```
HomeKit: Formatting HomeKit storage at 0x100000
```

gazodia, 2020-06-14:
Hey, sorry for the delay answering, I wasn't able to test anything with the esp this weekend. Now I'm back. 
So, I've tried exactly what you said, but when I flash and monitor at the same time something occurs that the monitor just doesn't work. Maybe I've missed something on the process:

```
make -C examples/led erase_flash
esptool.py -p /dev/tty.usbserial-1420 --baud 115200 erase_flash
esptool.py v2.8
Serial port /dev/tty.usbserial-1420
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:27:68:a5
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.4s
Hard resetting via RTS pin...

make -C examples/led flash monitor
esptool.py -p /dev/tty.usbserial-1420 --baud 115200 write_flash -fs 8m -fm qio -ff 40m \
		0x0 /Users/gabrielgazal/esp/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /Users/gabrielgazal/esp/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/led.bin 
WARNING: Flash size arguments in megabits like '8m' are deprecated.
Please use the equivalent size '1MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.8
Serial port /dev/tty.usbserial-1420
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:27:68:a5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0020
Compressed 3104 bytes to 2169...
Wrote 3104 bytes (2169 compressed) at 0x00000000 in 0.2 seconds (effective 114.9 kbit/s)...
Hash of data verified.
Compressed 2048 bytes to 23...
Wrote 2048 bytes (23 compressed) at 0x00001000 in 0.0 seconds (effective 1979.5 kbit/s)...
Hash of data verified.
Compressed 432896 bytes to 311253...
Wrote 432896 bytes (311253 compressed) at 0x00002000 in 30.1 seconds (effective 114.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
/Users/gabrielgazal/esp/esp-open-rtos//utils/filteroutput.py --port /dev/tty.usbserial-1420 --baud 115200 --elf ./build/led.out
Opening /dev/tty.usbserial-1420 at 115200bps...
```

After this the terminal only shows blank lines. Also I don't get any messages like yours, I've searched for it in the logs but nothing :( When was it supposed to show up?

gazodia, 2020-06-15:
@maximkulkin Got it to work!! I'm leaving my other comment here just for reference in case anyone needs it. The errors that appeared guided me to the solution :)

Basically I wasn't exporting the env variables in the right way. I'm using docker and have little knowledge in it, that tricked me a lot. Luckily enough, I stumbled across issue #185 after I got the errors I described in the previous comment ( https://github.com/maximkulkin/esp-homekit-demo/issues/113#issuecomment-643833592). Searching for the first line revealed the solution for the problem. Now I got the led example running on the Esp!!! It still works a little weirdly, but now I have some idea of what to do. 

I really want to thank you. I've been struggling for quite some time on this and finally got it to work. So happy about it. Thanks a lot @maximkulkin !!!