maximkulkin/esp-homekit-demo #114: Error reading data from socket (code 113)

konifer44, 2018-08-05:
What that error mean? Im using Sonoff switch app? It occurs about 30-60min after program start.

```
I (143392) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
!!! HomeKit: [Client 55] Error reading data from socket (code 113). Disconnecting
STA disconnected
>>> HomeKit: [Client 55] Closing client connection
```

kosukekurimoto, 2019-01-16:
I am experiencing the same problem.
```
!!! HomeKit: [Client 58] Error reading data from socket (code 113). Disconnecting
>>> HomeKit: [Client 58] Closing client connection
!!! HomeKit: [Client 60] Error reading data from socket (code 113). Disconnecting
>>> HomeKit: [Client 60] Closing client connection
>>> HomeKit: [Client 59] Closing client connection
```

LiangyongFShen, 2019-02-25:
@kosukekurimoto I have the same problem, have you solved it？

maximkulkin, 2019-04-27:
Is some functionality broken or you are just concerned about message in the log?

pbendersky, 2019-04-27:
I think this is what was happening when using `lwip` sockets. You solved a few weeks ago in the main library.

konifer44, 2019-07-09:
Seems to be fine now. Closing

closed 2019-07-09 by konifer44