maximkulkin/esp-homekit-demo #117: can it be used with arduino ide ?? having a lot of trouble compling and flashing using rtos ??

surya31, 2018-08-13:


maccoylton, 2018-08-13:
This guy was workign on an Arduino library https://github.com/An00bIS47/Homekit not sure how far he has progressed. 

Follow the step by step provided and let us know which part you are getting stuck at

surya31, 2018-08-16:
ok sir, the problem im facing is that the build version is different and cant use the basic examples of homekit whereas rtos examples are working for me. how do i compile the homekit code with new rtos version as this doesnt have rtos/bin folder so cant complie the code and flash

surya31, 2018-08-16:
evertime i "make" i het some errors or the other really confused now but really want to use standalone homekit switch

maximkulkin, 2018-08-16:
@surya31 well, there are no fortune tellers here, so if you do not give us details we won't be able to help.

I personally run on macOS and to workaround problem with toolchain not liking case insensitive file system I have all toolchain packaged into Docker containers. Here is my configuration:
* a docker image with esp-sdk
* a docker image with esp-open-rtos (based on esp-sdk image)
* esp-open-rtos repository
* esp-homekit-demo repository
then I just run build inside docker image (see commands example in Gist)

https://gist.github.com/maximkulkin/bca710444cbbc01a1845687a266f45a4

maximkulkin, 2018-08-16:
I just checked with latest esp-open-rtos version and it compiles without problems.