maximkulkin/esp-homekit-demo #119: Question for Air Conditioner accessory

peros550, 2018-08-16:
I am trying to control my wall mounted A/C using an ESP8266 board (Wemos D1 mini) and Homekit. I have already done this using Arduino SDK. 

Normally, my A/C is controlled through IR. For that I had to  add all IR data into a separate library. 

My questions is related to the homekit accessory. 

Normally an A/C accessory should have everything a thermostat has (Modes, Targets Temperatures) plus :
- Fan speed control (available Options: AUTO , LOW, MEDIUM, HIGH)
- Swing Control (Available options: ON / OFF) 

Do you know if there is a dedicated A/C accessory supported by Homekit? Or I should build it using a thermostat and other additional services (like for the fan control) 

Thanks in advance

renssies, 2018-08-16:
Yes, accessory category 21 is for Air Conditioners. I believe you only need 1 service for it: Heater Cooler. Somehow that service is still missing from esp-homekit but you should be able to implement it yourself quite easily. A heater cooler service has UDID `000000BC-0000-1000-8000-0026BB765291` and the following characteristics: 
- Name
- Active
- Current Heater Cooler State
- Target Heater Cooler State
- Current Temperature 
And optionally:
- Lock Physical Controls
- Swing mode 
- Cooling Threshold Temperature
- Heating Threshold Temperature
- Temperature DIsplay Units
- Rotation Speed

For Rotation Speed you would have to modify it. Standard this one is only in 0-100%, but you can modify it to have a custom unit of 0, 1, 2, 3. Where 0 is off, 3 is high. 

For auto speed is not supported by the Home app by default

renssies, 2018-08-16:
Since I noticed the new homekit accessories introduced in iOS 11.0 where completely missing I've decided to start adding them: https://github.com/renssies/esp-homekit/tree/heater-cooler-support. This includes the heater/cooler service and characteristics.

peros550, 2018-08-17:
@renssies Hi, many thanks for your efforts. It is very weird that there is no target temperature characteristic. Normally, all these AC units work like a thermostat having additional control elements for fan , swing  etc.  I am trying to build an empty example just to see what kind of UI the home app exposes for this accessory.

supersjimmie, 2018-08-17:
I don't understand way you say about the target temp.
The Thermostat has a characteristic ```TARGET_TEMPERATURE```.
How to set the heater and/or cooler to a target temperature?  Or do I have to manually read it's ```CURRENT_TEMPERATURE```, compare it to some desired value, and then switch on/off the heater/cooler by code?

peros550, 2018-08-20:
Is there any guide on how to interact with Homekit characteristics?  What iOS is expecting from our side and which actions on iOS needs to be expected on our program though call backs etc? 

I tried to build something based on thermostat example (as minimal as possible), but as soon as I register the accessory it seems to work  only for a few minutes, afterwards it give "no response" in home app. 

Here is my code for anyone who would like to have a look: 

```
/*
 * Thermostat example
 *
 * Wiring is as follows (requires ESP-12 as it needs 4 GPIOs):
 *
 * DHT11 (temperature sensor)
 *
 *               -------------
 *              |GND       VCC|    (These go to control pins of relay)
 *              |15         13| --> Heater
 *              |2          12| --> Cooler
 *              |0          14| --> Fan
 *              |5          16|
 * DHT Data <-- |4       CH_PD|
 *              |RXD       ADC|
 *              |TXD      REST|
 *               -------------
 *              |   |-| |-| | |
 *              | __| |_| |_| |
 *               -------------
 *
 */
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <espressif/esp_system.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <etstimer.h>
#include <esplibs/libmain.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"

#include <dht/dht.h>

#define LED_PIN 2
#define TEMPERATURE_SENSOR_PIN 4
#define FAN_PIN 14
#define COOLER_PIN 12
#define HEATER_PIN 13
#define TEMPERATURE_POLL_PERIOD 10000
#define HEATER_FAN_DELAY 30000
#define COOLER_FAN_DELAY 0

static void wifi_init() {
    struct sdk_station_config wifi_config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };

    sdk_wifi_set_opmode(STATION_MODE);
    sdk_wifi_station_set_config(&wifi_config);
    sdk_wifi_station_connect();
}

void thermostat_identify(homekit_value_t _value) {
    printf("Thermostat identify\n");
}

[... 95 lines cut ...]
            &heating_threshold,
            &units,
            &rotation_speed,
            NULL
        }),
        NULL
    }),
    NULL
};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void user_init(void) {
    uart_set_baud(0, 115200);

    wifi_init();
    thermostat_init();
    homekit_server_init(&config);
}

```
`

supersjimmie, 2018-08-20:
@peros550  not sure about yours, but I had such issues because too small memory sizes in the ```xTaskCreate``` calls.  After a while things will overflow.

maximkulkin, 2018-08-20:
@peros550 a debug log would really help. Mine thermostat runs on NodeMCU (from HiLetgo) no problem.

peros550, 2018-08-20:
Hi Maxim, I will provide logs later today. Once again many thanks for all this work you have put together.

maccoylton, 2018-08-20:
Can you also trying using the Eve App rather thanks the Home App, Eve gives some more detailed errors. For example if you have something wrong in your characteristics, Home might say not responding, but Eve will tell you the Accessory is blocked.

renssies, 2018-08-21:
To me, it feels like it's sending too many notifications but I'm not sure since I don't know over how big of a timespan this log is.

renssies, 2018-08-21:
Could you try and see what the LED example does? Some routers still do not play nicely with this implementation of mdns/bonjour. If the LED example has the same issue then it's most likely an issue we still have to fix but have no idea how to fix. 

What should work though is the ESP32, that one uses an "Apple Certified" mdns/bonjour implementation.

Kristian8606, 2018-08-21:
I still have no solution to the problem in a discussion 108

peros550, 2018-08-21:
All, I have some good news. I finally made it to have a working thermostat & A/C accessory. 
It looks that the command "git update submodule - init - recursive" did not properly bring any updates  and I was experiencing bugs that were already resolved. I had to completely remove the folder  "esp-homekit-demo" and download it again. 

@Kristian8606  are you still getting unresponsive accessory after a while? If yes, try the above step and let us know.

Kristian8606, 2018-08-22:
I had to reinstall my Macbook and install everything from the beginning and still the accessory does not respond. Do you think I have to delete the my folder and download it again?

Kristian8606, 2018-08-22:
would you share your code thermostat & A / C accessory.

peros550, 2018-08-22:
@Kristian8606 , in order to avoid such system wide changes I followed a guide that somebody posted on another thread. I installed a Ubuntu environment on a virtual machine. In this case, at least I don't have to reinstall the OS of my main computer if something goes wrong. 

Here are two links to get you started: 

1) https://www.youtube.com/watch?v=kYEzEFH5LiM
2) https://www.youtube.com/watch?v=T-oSjMCmNYk

In regards to the unresponsive situation, I believe there was a bug which was already resolved a few days ago, but the updated file was not present in my local repository. It turned out that command "git update submodule - init - recursive" did not bring updates from the git to my local folder. 

Upon deleting the whole esp-homekit-demo folder and running: 
1) "git clone https://github.com/maximkulkin/esp-homekit-demo"
and then navigate to esp-homekit-demo and run
2) "git update submodule - init - recursive"

seems to update all files. Then I ran the thermostat code with no changes from what you can find in Maxim's original implementation and worked well.

My A/C code is listed above. I still have not implemented any real logic, (only a few printf commands to interact with the UI and monitor  serial output)  but I intent to post the full code as soon as I have it ready. 

My next milestone is to try to send successfully IR commands through an IR LED attached to the ESP8266.

Kristian8606, 2018-08-22:
Hello I deleted the folder and then I cloned to another directory and now the thermostat works. I had done these steps before, but only when I changed the directory the thermostat started working 
thank you!

pbendersky, 2018-11-21:
@peros550 I've been working on something similar for the past months, but can't get IR codes to be sent reliably. Were you able to?

peros550, 2018-11-24:
@pbendersky Hi Pablo. I did a few preliminary attempts but I wasn't successful and then I didn't have time to devote on this goal since my last message here in this thread. 

In the past I have sent successfully IR commands to my A/C but that was using Arduino SDK and IRremoteESP8266  library. In order to do that I used the AnalysIR program which helped me capture correctly the timings of the pulses sent by the original remote controller. Then using the command sendraw("_raw data of the IR command_ ") I was able to send commands back to the A/C using the Arduino SDK. 

What's the issue you are facing at your side? Have you sent any command successfully or you are constantly failing?

NS-K, 2018-11-26:
@peros550 @pbendersky Hi,
I am in the same situation. i am able to send the ir signal but it work only 1 of 3 times.
I ve tried sending via xCreateTask with 2 as priority and also directly from a sendRaw function.
the problem is that sometimes( 1 of 3 try) the signal sent by my ESP have different timings than the original

pbendersky, 2018-11-26:
@peros550 I'm using [this IR library](https://github.com/slawekkolodziej/irremote), and as @NS-K said, it works only about 1/3 or 1/4 of the times I send an IR.

From what I've been reading, it would be more reliable to send IR using I2S or the serial port to modulate the 38khz carrier signal, but I haven't been able to try it yet.

NS-K, 2018-11-26:
i am using the same library
Even overclocking the CPU didn't help in my situation.
i ve started playing today with an YS-IRTM (SERIAL / IR) but it's only able to send NEC protocol.
ANY NEW IDEA ??

pbendersky, 2018-11-26:
Not yet... My next step is to try using I2S to modulate, but haven't had the chance yet.

pbendersky, 2018-12-25:
@NS-K @peros550 I just uploaded a IR library that's way more reliable for me. It uses a trick described in AnalysIR to modulate IR using the secondary UART on an ESP8266.

You can check it out here: https://github.com/pbendersky/ir-esp-uart

peros550, 2018-12-25:
Thank you! 

My experience is minimum and I can't really follow your suggestion on how to migrate a command like this: 

`
uint16_t Cool_18_III[] = {8944, 4344, 600, 1700, 600,....more numbers comings AnalysIR....};
 irsend.sendRaw(Cool_18_III, sizeof(Cool_18_III) / sizeof(uint16_t), 38);`

If you could provide an example it would be very helpful for me. 

Many thanks

pbendersky, 2018-12-27:
If you look at the code within `sendRaw`, you'll probably notice it's a series of `mark` / `space` calls.

peros550, 2018-12-27:
@pbendersky Thank you! Your code works nicely!

closed 2019-01-20 by peros550