maximkulkin/esp-homekit-demo #120: how do i pair it with homekit . installed the files and running on my esp but cant find the device on homekit

surya31, 2018-08-17:


royfocker19, 2018-08-17:
Could you please provide more info on the steps you did? 
Did you connect to the generated AP on the eso to connect to your WiFi?

royfocker19, 2018-08-17:
> i just flashed the examples/button and connected to wifi but cant see the device on homekit

You don’t see any accessory when pressing + “add accessory”?

Press in Dire Situations http://www.nooooooooooooooo.com/

surya31, 2018-08-17:
i saw this once but when deleted , it not showing at all, board is fine, monitor also displays its connected but no homkit authentication is happening

idumzaes, 2018-08-17:
Post you code...

surya31, 2018-08-17:
m_task_hdl : 3ffef8e0, prio:1, stack:176
frc2_time:0x3fff3f28, prio:12, stack:200
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: F4:4E:D1:41:E9:A2
>>> HomeKit: Found admin pairing with 23E9459B-46A6-44E1-B35C-8E71AFAA324C, disabling pair setup
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Dual Lamp 
                                  md=Dual Lamppv=1.0id=F4:4E:D1:41:E9:A2c#=1s#=1ff=0sf=0ci=5 Port=5556 TTL=4500
scandone
add 0
aid 1
cnt

connected with hell000, channel 6
ip:192.168.69.69,mask:255.255.255.0,gw:192.168.69.3

surya31, 2018-08-17:
iam using sonoff_dual from example earlier it was running now i cant find it. plus is there a way to change accessory id ??

surya31, 2018-08-17:
iam a noob in iot just got started with esp

dezdeepblue, 2018-08-18:
It is already paired, so if you've deleted the accessory in the Home app you should erase flash and reprogram the example:
`make -C examples/button erase_flash`
`make -C examples/button flash`
Then add it again in the home app.

surya31, 2018-08-18:
so this chip get stored to the home api?? as i tried flashing other codes as well still no lock in fiding the accessory ?I tried this.. still no luck

dezdeepblue, 2018-08-18:
Could you successfully erase? I think you should give more details about what happened after making and then after flashing. Post the output as you did

surya31, 2018-08-18:
yes sir i erased and evertything is working 
bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/sonoff_dual_lights.bin 
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0320
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (86.9 kbit/s)...
Writing 4096 @ 0x1000... 4096 (100 %)
Wrote 4096 bytes at 0x1000 in 0.4 seconds (88.5 kbit/s)...
Writing 430080 @ 0x2000... 430080 (100 %)
Wrote 430080 bytes at 0x2000 in 37.3 seconds (92.1 kbit/s)...
Leaving...
make: Leaving directory '/home/surya/Development/esp-open-sdk/esp-open-rtos/examples/sonoff_dual_lights'

surya31, 2018-08-18:
ESP-Open-SDK ver: 0.0.1 compiled @ Aug 17 2018 08:09:50
phy ver: 273, pp ver: 8.3

mode : sta(18:fe:34:cf:0f:75)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 66:14:29:7F:4D:17
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Dual Lamp 
                                  md=Dual Lamppv=1.0id=66:14:29:7F:4D:17c#=1s#=1ff=0sf=1ci=5 Port=5556 TTL=4500
scandone
add 0
aid 3
cnt

connected with nav52, channel 1
ip:192.168.69.69,mask:255.255.255.0,gw:192.168.69.3

surya31, 2018-08-18:
earlier here a authentication homekit used to come but after deleting the accessory it not showing again

surya31, 2018-08-18:
the problem is it was showing beofre was working fine but suddenly after last delete its gone forever

surya31, 2018-08-18:
de : sta(18:fe:34:cf:0f:75)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 9B:97:A9:E2:32:33
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Dual Lamp-CF0F75 
                                         md=Dual Lamppv=1.0id=9B:97:A9:E2:32:33c#=1s#=1ff=0sf=1ci=5 Port=5556 TTL=4500
scandone
add 0
aid 3
cnt

connected with nav52, channel 1
dhcp client start...
ip:192.168.1.109,mask:255.255.255.0,gw:192.168.1.1

surya31, 2018-08-18:
found another ios device and its showing here .. thank you guys for the support

royfocker19, 2018-08-18:
You should also close home app and open it again, not just restart your phone.

maximkulkin, 2018-08-18:
@surya31 When you remove accessory from iPhone, it is better to do it while accessory is online and accessible. When you remove accessory, iPhone send command to remove itself from accessory's pairing database. If you remove it offline, the pairing record will remain in accessory's database and it won't pair again. In that case you need to erase pairing database from accessory's flash. Not going into much detail, it is easier to just erase_flash and flash firmware again.

closed 2019-01-27 by maximkulkin