maximkulkin/esp-homekit-demo #124: No response after ESP8266 Reboot

bwicks, 2018-08-19:
Hi all,
I've flashed the example led to my ESP8266 however when I move it from my USB flasher and power it back on I'm left with the accessory unable to respond. Am I doing something wrong?

maximkulkin, 2018-08-20:
Logs please

konifer44, 2018-08-20:
I have similar issue, accessory is not responding after few minutes after start.

buschco, 2018-09-07:
Is there a fix? I/we can't get any logs because the ESP won't response at all.

berikv, 2018-11-20:
Hi @maximkulkin,

I also ran into this problem, on the WeMos D1 Mini (4MB) with ESP8266EX chip.
Once the pairing is broken, the only way to recover is to run `homekit_storage_reset()` and redo the pairing.
My impression is that the pairing breaks when Homekit tries to control the ESP while the homekit-esp library is busy initialising on the wifi connection: Most of the power-cycles maintain the pairing, but when the home app is opened while the esp is booting, pairing breaks. 

Here are some logs. Let me know if I can help debug further.

Power-cycle with successful pairing
```
ESP-Open-SDK ver: 0.0.1 compiled @ Nov 17 2018 18:43:34
phy ver: 273, pp ver: 8.3

init start
init complete..
mode : sta(dc:4f:22:55:cf:53)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 2D:DD:27:0C:2B:03
>>> HomeKit: Found admin pairing with 9239412C-F341-4B65-BFC4-A559D7DE5DC6, disabling pair setup
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
scandone
add 0
aid 2
cnt

connected with my-wifi, channel 11
dhcp client start...
ip:192.168.2.15,mask:255.255.255.0,gw:192.168.2.254
mDNS announcement: Name=Window blind-2DDD 
                                          md=MyCurtainpv=1.0id=2D:DD:27:0C:2B:03c#=1s#=1ff=0sf=0ci=14 Port=5556 TTL=4500
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with 9239412C-F341-4B65-BFC4-A559D7DE5DC6
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: [Client 4] Get Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
```

Power-cycle with unsuccessful pairing
```
ESP-Open-SDK ver: 0.0.1 compiled @ Nov 17 2018 18:43:34
phy ver: 273, pp ver: 8.3

init start
init complete..
mode : sta(dc:4f:22:55:cf:53)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 2D:DD:27:0C:2B:03
>>> HomeKit: Found admin pairing with 9239412C-F341-4B65-BFC4-A559D7DE5DC6, disabling pair setup
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
scandone
add 0
aid 2
cnt

connected with my-wifi, channel 11
dhcp client start...
ip:192.168.2.15,mask:255.255.255.0,gw:192.168.2.254
mDNS announcement: Name=Window blind-2DDD 
                                          md=MyCurtainpv=1.0id=2D:DD:27:0C:2B:03c#=1s#=1ff=0sf=0ci=14 Port=5556 TTL=4500
```

When the pairing failed once, it will fail every time
```
ESP-Open-SDK ver: 0.0.1 compiled @ Nov 17 2018 18:43:34
phy ver: 273, pp ver: 8.3

init start
init complete..
mode : sta(dc:4f:22:55:cf:53)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 2D:DD:27:0C:2B:03
>>> HomeKit: Found admin pairing with 9239412C-F341-4B65-BFC4-A559D7DE5DC6, disabling pair setup
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
scandone
add 0
aid 2
cnt

connected with my-wifi, channel 11
dhcp client start...
ip:192.168.2.15,mask:255.255.255.0,gw:192.168.2.254
mDNS announcement: Name=Window blind-2DDD 
                                          md=MyCurtainpv=1.0id=2D:DD:27:0C:2B:03c#=1s#=1ff=0sf=0ci=14 Port=5556 TTL=4500
```

hfosteriv, 2019-02-17:
I’m having a similar issue, I’m unable to reestablish a connection even if I make sure that the HomeKit app is closed completely before/during restart. @berikv were you able to implement homekit_storage_reset() upon a failed pairing? I’m not even able to monitor logs after a power cycle (my board has a switch so I’m not actually disconnecting the UCB cable). 

I’d be interested in talking out some ideas to remedy this issue. My biggest concern (untested) is when my HomePod gets here it will be my homehub and if there is ever a power failure I’d imagine that the homehub would attempt to establish that connection immediately upon restored power, so in any instance of what you described (berkiv) the pairing would always fail. 

Would a simple delay after a reboot solve this?

hfosteriv, 2019-02-19:
Check out my related issue/discussion I just posted, I am thinking this is hardware difference related. Issue #171

maximkulkin, 2019-02-26:
Check your board documentation: it might be like with Spark Fun ESP8266 dev board you need to pull some pin high/low to have it boot program instead of bootloader. See #171

closed 2019-02-26 by maximkulkin