maximkulkin/esp-homekit-demo #126: No Response with multiple esp8266 connected

bryantcoles, 2018-08-22:
I am using the temperature sensor code and have successfully added it to my Homekit Setup. Have tested for two weeks, and It works very well. If I add any other esp8266 to HomeKit using the temperature sensor code it appears that only one sensor will communicate with the others “Not Responding” if I wait a few minutes and go back and refresh, another sensor will be respond, with the others “Not Responding” Any ideas what’s happening? I’ve tried changing the serial numbers and pair codes with no luck.

bryantcoles, 2018-08-24:
Here is a snippet from my log...I have done a complete new install and the esp8266 stalls at this state not able to connect to MyHome.app. As I mentioned before, this worked perfectly a week or so ago. 

>>> HomeKit: Generated new accessory ID: 3D:39:9C:71:69:E9
>>> HomeKit: Generated new accessory key
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Temperature Sensor md=MyTemperatureSensorpv=1.0id=3D:39:9C:71:69:E9c#=1s#=1ff=0sf=1ci=9 Port=5556 TTL=4500
>>> homekit_run_server: Staring HTTP server
scandone
add 0
aid 7
cnt

connected with Unifi, channel 1
dhcp client start...
beacon timeout
rm match
scandone
add 0
aid 4
cnt

connected with Unifi, channel 11
dhcp client start...
ip:192.168.1.110,mask:255.255.255.0,gw:192.168.1.1

HomeACcessoryKid, 2018-08-26:
see for this pull request for my solution

[maximkulkin/esp-homekit] Bonjour improvements (#60)

bryantcoles, 2018-08-26:
implementing the fix in issue 60 per suggestion above does not resolve the issue. 

esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0320
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.8 kbit/s)...
Writing 4096 @ 0x1000... 4096 (100 %)
Wrote 4096 bytes at 0x1000 in 0.4 seconds (89.8 kbit/s)...
Writing 434176 @ 0x2000... 434176 (100 %)
Wrote 434176 bytes at 0x2000 in 37.8 seconds (92.0 kbit/s)...
Leaving...
/Volumes/case-sensitive/esp-open-rtos///utils/filteroutput.py --port /dev/cu.SLAB_USBtoUART --baud 115200 --elf ./build/temperature_sensor.out
Opening /dev/cu.SLAB_USBtoUART at 115200bps...
mode : sta(60:01:94:75:51:1f)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x7a000
>>> HomeKit: Generated new accessory ID: 78:51:F6:6D:F1:35
scandone
>>> HomeKit: Generated new accessory key
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
add 0
aid 7
cnt
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Temperature Sensor md=MyTemperatureSensorpv=1.0id=78:51:F6:6D:F1:35c#=1s#=1ff=0sf=1ci=9 Port=5556 TTL=4500
>>> homekit_run_server: Staring HTTP server

connected with Unifi, channel 1
dhcp client start...
beacon timeout
rm match
scandone
add 0
aid 7
cnt

connected with Unifi, channel 11
dhcp client start...
ip:192.168.1.132,mask:255.255.255.0,gw:192.168.1.1

HomeACcessoryKid, 2018-08-26:
you must not have applied it properly since the name did not get the 4 digits of the ID appended

bryantcoles, 2018-08-27:
I really appreciate all the help here. I still have unresponsive accessories after I add them to MyHome. I have followed all suggested solutions with no luck. Here is my make file and logs. Any other ideas?

Makefile

PROGRAM = temperature_sensor

EXTRA_COMPONENTS = \
    extras/dht \
    extras/http-parser \
    $(abspath ../../components/wolfssl) \
    $(abspath ../../components/cJSON) \
    $(abspath ../../components/homekit)

# DHT11 sensor pin
SENSOR_PIN ?= 4

FLASH_SIZE ?= 4

EXTRA_CFLAGS += -I../.. -DHOMEKIT_SHORT_APPLE_UUIDS -DSENSOR_PIN=$(SENSOR_PIN)

include $(SDK_PATH)/common.mk

ESPTOOL_ARGS=-fs $(FLASH_SIZE)MB -fm $(FLASH_MODE) -ff $(FLASH_SPEED)m 

monitor:
    $(FILTEROUTPUT) --port $(ESPPORT) --baud 115200 --elf $(PROGRAM_OUT)

LOGS

LD build/temperature_sensor.out
FW firmware/temperature_sensor.bin
esptool.py v2.5.1-dev
Creating image for ESP8266...
esptool.py -p /dev/cu.SLAB_USBtoUART --baud 115200 erase_flash
esptool.py v2.5.1-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 60:01:94:75:51:1f
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.0s
Hard resetting via RTS pin...
esptool.py -p /dev/cu.SLAB_USBtoUART --baud 115200 write_flash -fs 4MB -fm dout -ff 40m  \
		0x0 /Volumes/case-sensitive/esp-open-rtos//bootloader/firmware_prebuilt/rboot.bin 0x1000 /Volumes/case-sensitive/esp-open-rtos//bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/temperature_sensor.bin 
esptool.py v2.5.1-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 60:01:94:75:51:1f
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0340
Compressed 3104 bytes to 2169...
Wrote 3104 bytes (2169 compressed) at 0x00000000 in 0.2 seconds (effective 126.2 kbit/s)...
Hash of data verified.
Compressed 2048 bytes to 23...
Wrote 2048 bytes (23 compressed) at 0x00001000 in 0.0 seconds (effective 2361.5 kbit/s)...
Hash of data verified.
Compressed 433060 bytes to 309849...
Wrote 433060 bytes (309849 compressed) at 0x00002000 in 27.4 seconds (effective 126.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
/Volumes/case-sensitive/esp-open-rtos///utils/filteroutput.py --port /dev/cu.SLAB_USBtoUART --baud 115200 --elf ./build/temperature_sensor.out
Opening /dev/cu.SLAB_USBtoUART at 115200bps...
mode : sta(60:01:94:75:51:1f)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x100000
>>> HomeKit: Generated new accessory ID: DF:BB:DA:05:1B:B8
>>> HomeKit: Generated new accessory key
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
Function called without core lock
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Temperature Sensor md=MyTemperatureSensorpv=1.0id=DF:BB:DA:05:1B:B8c#=1s#=1ff=0sf=1ci=9 Port=5556 TTL=4500
>>> homekit_run_server: Staring HTTP server
scandone
add 0
aid 7
cnt

connected with Unifi, channel 11
dhcp client start...
ip:192.168.1.132,mask:255.255.255.0,gw:192.168.1.1

HomeACcessoryKid, 2018-09-01:
the fact that MDNS still uses "Name=Temperature Sensor“ instead of "Name=Temperature Sensor-DFBB” shows the fix is not yet applied

Please check out if you code is really updated…

Good luck, HacK

peros550, 2018-12-25:
Hey @HomeACcessoryKid  I'm having the same issue. I'm using a Wemos D1 mini device which has 4MB flash. 
In my case the fix would be something like that ?

```
FLASH_SIZE ?= 4
# firmware tool arguments
ESPTOOL_ARGS=-fs $(FLASH_SIZE)MB -fm $(FLASH_MODE) -ff $(FLASH_SPEED)m 
```

cottsak, 2019-02-16:
I also encountered the same issue with a D1 mini

maximkulkin, 2019-04-22:
@cottsak As far as I understand, original issue was because multiple accessories had same name and thus mDNS collisions. You should use unique names for all accessories. One way to do that is to append mac address at the end of accessory name. Most examples already do that.

So, what's your issue now?

peros550, 2019-05-08:
@maximkulkin 

If you have some time, please check the implementation of a watchdog timer in mdnsresponder.c here : 

https://github.com/RavenSystem/esp-homekit-devices/commit/f5f642fc0390047ed0600a00ac91df215eaf8c50

It may help to quickly re-announce accessories in case of any issue. What do you think?

maximkulkin, 2019-05-12:
@peros550 I've added a recurring (every 2 seconds) check for IP address and reannouncement.

closed 2019-05-12 by maximkulkin