maximkulkin/esp-homekit-demo #129: MQTT example

fanlessfan, 2018-09-04:
Could any one add an MQTT example? so that the homekit sensor can report to smart home controller like openhab.

maximkulkin, 2018-09-04:
It sounds like it will just be a different type of firmware - the one that integrates over MQTT. The purpose of this framework is to integrate specifically with HomeKit.

fanlessfan, 2018-09-04:
you are right. but MQTT is light weight and flexible protocol. I tried to combine two together, but not working so well. Just hope someone knows deeper can provide a example. there is no need to add it in the esp-homekit library. I saw someone has the same request in esp-homekit issues. https://github.com/maximkulkin/esp-homekit/issues/49  just want to raise the attention to someone who is good at it.

Thanks again for your great work on esp-homekit

idumzaes, 2018-09-04:
I’d be interested in adding MQTT publish messages to this firmware. I struggled to get a button integrated with the LED Strip example but I may be up for the challenge if anyone else is willing to help out.

closed 2019-01-27 by maximkulkin

lrusnac, 2020-01-19:
if people are still interested I have an example here https://github.com/lrusnac/esp-homekit-mqtt-demo/blob/master/led_mqtt/main.c

fanlessfan, 2020-01-20:
👍