maximkulkin/esp-homekit-demo #130: esp won't connect to wifi after restarting it (wifi_config example)

buschco, 2018-09-05:
### Problem
Connecting to WIFI, pairing to HomeKit and controlling the LED works fine but **after disconnecting the ESP from the power it will not connect to the previously configured WIFI.**
### Attempts
I tried two different WIFI networks and two different ESP-8266 (the small one with 8 pins).
### Logs
I was not able to get the anything from the ESP. Probably my 8-Pin ESP to USB converter does not support debugging.

maccoylton, 2018-09-07:
if you can flash, you should be able to us the same connection to see the trace .. did you compile with debugging turned on in the Makefile?

buschco, 2018-09-07:
I was able to get logs now. It seems after restarting the ESP it would stop working.
`Opening /dev/tty.wchusbserial1430 at 115200bps...` then nothing. I think this issue is a duplicate of #124

hampusiggstrom, 2018-10-02:
I got the same problem seems to be related to the 8 pin version of ESP8622-01. ESP-12F works just fine for me.

hfosteriv, 2019-02-19:
Check out my related issue/discussion I just posted, I am thinking this is hardware difference related. Issue #171

maximkulkin, 2019-02-26:
Please check documentation on your board, it might be that you need to pull some pin high/low to force it loading program instead of bootloader after power reset.

closed 2019-02-26 by maximkulkin