maximkulkin/esp-homekit-demo #132: Accessory not responding after some time

MirceaIs, 2018-09-19:
Hello,

First of all, thank you to all contributors for sharing this library.
 
my setup:
latest version of homekit and dependinces cloned
M5Stack board / ESP32
iPhone X
tried two different setups: thermostat and temperature sensor.

I'm able to establish the pairing and install the accessory in my Home application, in a few minutes the accessory is not responding. After accessory restart all works fine for a few more minutes.
My implementation is similar with the one in the temperature sensor example, see below:

homekit_characteristic_t temperature = HOMEKIT_CHARACTERISTIC_(CURRENT_TEMPERATURE, 0);
homekit_characteristic_t humidity    = HOMEKIT_CHARACTERISTIC_(CURRENT_RELATIVE_HUMIDITY, 0);

void thermostat_identify(homekit_value_t _value) {
    ESP_LOGI("HOMEKIT","Thermostat identify");
}

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_thermostat, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Thermo"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Mir"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "001"),
            HOMEKIT_CHARACTERISTIC(MODEL, "MyThermo"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, thermostat_identify),
            NULL
        }),
        HOMEKIT_SERVICE(TEMPERATURE_SENSOR, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Temperature Sensor"),
            &temperature,
            NULL
        }),
        HOMEKIT_SERVICE(HUMIDITY_SENSOR, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Humidity Sensor"),
            &humidity,
            NULL
        }),
        NULL
    }),
    NULL
};

homekit_server_config_t homekit_config = {
    .accessories = accessories,
    .password = "111-11-111"
};

homekit_config is sent as parameter to the homekit_server_init function called in SYSTEM_EVENT_STA_GOT_IP event. temperature and humidity characteristics are updates every 30s in a different thread. 

I wonder what am I doing wrong?

maximkulkin, 2018-09-19:
There are people reporting that problem but I can not reproduce it on any of my devices (I use NodeMCU devices from HiLetgo and Makerfocus and a Sonoff Dual R2). If you could collect and post debug logs (with HOMEKIT_DEBUG=1) of devices when they become unavailable that could help nailing that issue.

MirceaIs, 2018-09-20:
Hello,

Below the debug logs of my accessory device.
Between the last two lines in the log there is a time interval of up to a minute or longer. First I get the "not responding" message in Home app at random timing (range of minutes) after the last "Finished processing" line appears in console. "Closing client connection" massage comes about 1min after "not responding" in the app. 

I would expect that after "Closing client connection" message, a Home app restart resumes the connection but it doesn't.

I also disabled any other tasks to avoid race conditions or memory problems.

see log below:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:4848
ho 0 tail 12 room 4
load:0x40078000,len:8000
load:0x40080400,len:5548
entry 0x400806d4
I (182) boot: ESP-IDF v3.2-dev-773-gb999d10e-dirty 2nd stage bootloader
I (183) boot: compile time 19:35:27
I (188) boot: Enabling RNG early entropy source...
I (205) boot: SPI Speed      : 40MHz
I (218) boot: SPI Mode       : DIO
I (230) boot: SPI Flash Size : 4MB
I (243) boot: Partition Table:
I (254) boot: ## Label            Usage          Type ST Offset   Length
I (277) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (300) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (324) boot:  2 factory          factory app      00 00 00010000 00100000
I (347) boot: End of partition table
I (360) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x181e0 ( 98784) map
I (595) esp_image: segment 1: paddr=0x00028208 vaddr=0x3ffb0000 size=0x031a8 ( 12712) load
I (625) esp_image: segment 2: paddr=0x0002b3b8 vaddr=0x3ffb31a8 size=0x00000 (0) load
I (627) esp_image: segment 3: paddr=0x0002b3c0 vaddr=0x40080000 size=0x00400 (1024) load
I (653) esp_image: segment 4: paddr=0x0002b7c8 vaddr=0x40080400 size=0x04848 ( 18504) load
I (722) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x7bed0 (507600) map
I (1784) esp_image: segment 6: paddr=0x000abef0 vaddr=0x40084c48 size=0x0a53c (42300) load
I (1888) esp_image: segment 7: paddr=0x000b6434 vaddr=0x400c0000 size=0x00000 (0) load
I (1890) esp_image: segment 8: paddr=0x000b643c vaddr=0x50000000 size=0x00000 (0) load
I (1966) boot: Loaded app from partition at offset 0x10000
I (1966) boot: Disabling RNG early entropy source...
I (1968) cpu_start: Pro cpu up.
I (1979) cpu_start: Starting app cpu, entry point is 0x40080e2c
I (1) cpu_start: App cpu up.
I (2012) heap_init: Initializing. RAM available for dynamic allocation:
I (2033) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (2052) heap_init: At 3FFB9988 len 00026678 (153 KiB): DRAM
I (2071) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (2091) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (2111) heap_init: At 4008F184 len 00010E7C (67 KiB): IRAM
I (2130) cpu_start: Pro cpu start user code
I (39) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (40) MAIN: Hello Thermostat!
I (90) wifi: wifi driver task: 3ffc1228, prio:23, stack:3584, core=0
I (90) wifi: wifi firmware version: ece97c4
I (90) wifi: config NVS flash: enabled
I (90) wifi: config nano formating: disabled
I (90) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (100) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (130) wifi: Init dynamic tx buffer num: 32
I (130) wifi: Init data frame dynamic rx buffer num: 32
I (130) wifi: Init management frame dynamic rx buffer num: 32
I (140) wifi: Init static rx buffer size: 1600
I (140) wifi: Init static rx buffer num: 10
[... 98 lines cut ...]
        "ev":   true
}
>>> homekit_server_on_update_characteristics: [Client 55] There were no processing errors, sending No Content response
>>> client_send: [Client 55] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 55] Finished processing
>>> homekit_client_process: [Client 55] Got 216 incomming data
>>> homekit_client_process: [Client 55] Decrypting data
>>> homekit_client_process: [Client 55] Decrypted 198 bytes, available 0
Decrypted data (198 bytes): "PUT /characteristics HTTP/1.1\x0D\x0AHost: Thermo._hap._tcp.local\x0D\x0AContent-Length: 79\x0D\x0AContent-Type: application/hap+json\x0D\x0A\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"ev":true},{"aid":1,"iid":13,"ev":true}]}"
>>> HomeKit: [Client 55] Update Characteristics
>>> homekit_server_on_update_characteristics: Free heap: 218624
>>> homekit_server_on_update_characteristics: [Client 55] Processing element {
        "aid":  1,
        "iid":  10,
        "ev":   true
}
>>> homekit_server_on_update_characteristics: [Client 55] Processing element {
        "aid":  1,
        "iid":  13,
        "ev":   true
}
>>> homekit_server_on_update_characteristics: [Client 55] There were no processing errors, sending No Content response
>>> client_send: [Client 55] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 55] Finished processing
>>> HomeKit: [Client 55] Closing client connection

ziolelle, 2018-11-22:
Hi, the same for me, but sorry I have no log... 
I will investigate little more

MirceaIs, 2018-11-22:
Hello,

Just a short update...
I've found some problems with the mdns client I was using. The mdns implementation in the esp-idf master branch is buggy and it stops responding after a while. This might explain why my hk accessory wasn't responding. I compiled against a clean copy of IDF release v3.2 and now the mdns problem is solved. However i still get sporadically device not responding in the current implementation event with mdns responding. There seems to be one additional issue which I'll try to debug when I'll find some time.

ziolelle, 2018-11-27:
In my experience, it seems that it only on ESP32 implementation that it fails, the same (almost...) code for homekit project under ESP8266 implmentation doesn't suffer of this.
With last changes in EspressIF SDK the think is better but also for me, sometime the ESP32 is not responding.

peros550, 2018-12-25:
I have Wemos D1 mini failing as well. Has anybody found any solution?

peros550, 2018-12-26:
> I compiled against a clean copy of IDF release v3.2 and now the mdns problem is solved. 

Could you please provide some instructions how did you do that? I would like to give it a try. 

Thanks

MirceaIs, 2018-12-26:
Just clone esp-idf from release v3.2 branch here: 
https://github.com/espressif/esp-idf/tree/release/v3.2
This branch should contain a stable implementation of MDNS responder.
Make sure to compile your project against this version by updating IDF–PATH in your build environment.

kvanbael, 2018-12-28:
I have the same issue: ESP32, homekit 'not responding' after a few hours.
Thanks for the tip to try the ESP-IDF release/v3.2... but that doesn't seem to solve my problem.

kvanbael, 2019-01-02:
[Here](https://www.dropbox.com/s/ramog08qk2cw1el/esplog.txt?dl=0)'s a log of the issue with homekit debug info (ESP-IDF v3.2, ESP32 chip, several apple devices).

At the end of the log, at least 2 apple devices were showing the accessory as "not responding". Refreshing the Home app (bringing it to the background and foreground) didn't trigger any new log output.

I don't see anything suspicious on the log, except for the connections being closed. It seems like the issue is at another layer (mDNS?)

`>>> HomeKit: [Client 57] Closing client connection`
`>>> HomeKit: [Client 59] Closing client connection`

bryantcoles, 2019-01-04:
I have the exact same issue using the Temp Sensor example with a ESP8266-12E and  DHT22. Even with these problems, the work that has been done here is excellent. Is anyone aware of any other diy HomeKit development environments that we can test against?

peros550, 2019-01-04:
> I have the exact same issue using the Temp Sensor example with a ESP8266-12E and DHT22. Even with these problems, the work that has been done here is excellent. Is anyone aware of any other diy HomeKit development environments that we can test against?

Sure thing this is an excellent work. 

In regards to your questions, have a look in this also excellent repo>> https://github.com/RavenSystem/esp-homekit-devices

There is also a similar issue discussed here: https://github.com/RavenSystem/esp-homekit-devices/issues/130

I don't know if this issue is caused by the code being not very well optimized or because of another layer not performing as expected (i.e. TCP, mDNSs etc) or even because of bad wifi environment or settings.

ziolelle, 2019-01-04:
> I have the exact same issue using the Temp Sensor example with a ESP8266-12E and DHT22. Even with these problems, the work that has been done here is excellent. Is anyone aware of any other diy HomeKit development environments that we can test against?

There is also this repository, I didn't test it deeply
https://github.com/younghyunjo/esp32-homekit

closed 2019-01-27 by maximkulkin