maximkulkin/esp-homekit-demo #133: ifan02 ceiling fan controller?

vmarks, 2018-10-05:
Hi, 
I'd like to use this to flash the sonoff ifan02 ceiling fan controller. Ideally, it would control 3 rotation speeds and the light on/off/dimming. 

Is this possible? What changes need to be made to the firmware for this to work with HomeKit?

vmarks, 2018-10-27:
checking in. Can I send a developer the ifan02 to help foster development?

idumzaes, 2018-10-27:
I’d love to get my hands on one of those as I essentially built one myself with an esp8266 and a dual relay module. Granted that my build doesn’t control the fan speed, just on/off state.

vmarks, 2018-10-27:
> I’d love to get my hands on one of those as I essentially built one myself with an esp8266 and a dual relay module. Granted that my build doesn’t control the fan speed, just on/off state.

Does yours show up in HomeKit? Want to work on making the iFan02 work in HomeKit? I was attempting to use one with Tasmota firmware and then use it via MQTT in Home Assistant, controlling that from HomeKit, but that's way more involved than I'd actually like it to be - I'd like to just install a firmware and have it be HomeKit capable.

idumzaes, 2018-10-27:
I use the Tasmota firmware with MQTT. I then use Node-Red to bridge it to HomeKit.
I wouldn’t mind working on one of those. If you’d like, I can even share my node-red flows with you. Are you willing to send me one?

vmarks, 2018-10-27:
Interesting! So you have a MQTT broker running and Node-Red, and what else?

It isn't exactly what I was hoping for - I hoped we could get support for iFan02 directly in this HomeKit firmware - but it sounds potentially lighter than HomeAssistant.

idumzaes, 2018-10-27:
Yes, I have a broker and NodeRed running on an original Pine64 1GB Ram board. I like to use NodeRed and MQTT because I can integrate multiple devices I have built, like switches and sensors.
But if you wanted to skip the broker and NodeRed, you could probably get away with using the Sonoff_Dual example from here.

vmarks, 2019-01-26:
{ "Sonoff iFan02", // Sonoff iFan02 (ESP8285)
GPIO_KEY1, // GPIO00 WIFI_KEY0 Virtual button 1 as feedback from RC
GPIO_USER, // GPIO01 ESP_TXD Serial RXD and Optional sensor
0, // GPIO02 ESP_LOG
GPIO_USER, // GPIO03 ESP_RXD Serial TXD and Optional sensor
GPIO_REL3, // GPIO04 WIFI_O2 Relay 3 (0 = Off, 1 = On) controlling the fan
GPIO_REL2, // GPIO05 WIFI_O1 Relay 2 (0 = Off, 1 = On) controlling the fan
// GPIO06 (SD_CLK Flash)
// GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
// GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
GPIO_KEY2, // GPIO09 WIFI_KEY1 Virtual button 2 as feedback from RC
GPIO_KEY3, // GPIO10 WIFI_KEY2 Virtual button 3 as feedback from RC
// GPIO11 (SD_CMD Flash)
GPIO_REL1, // GPIO12 WIFI_O0 Relay 1 (0 = Off, 1 = On) controlling the light
GPIO_LED1_INV, // GPIO13 WIFI_CHK Blue Led on PCA (0 = On, 1 = Off) - Link and Power status
GPIO_KEY4, // GPIO14 WIFI_KEY3 Virtual button 4 as feedback from RC
GPIO_REL4, // GPIO15 WIFI_O3 Relay 4 (0 = Off, 1 = On) controlling the fan
0, 0
},

Off -> Relay 2: OFF; Relay 3: OFF; Relay 4: OFF
Low -> Relay 2: ON; Relay 3: OFF; Relay 4: OFF
Med -> Relay 2: ON; Relay 3: ON; Relay 4: OFF
High -> Relay 2: OFF; Relay 3: OFF; Relay 4: ON

Starter -> Always set to High for 3 seconds.

I'm not sure how this should be integrated in the firmware, but it seems like the information that is needed, @maximkulkin

closed 2019-02-16 by maximkulkin