maximkulkin/esp-homekit-demo #135: Тemperature sensor on battery

Kristian8606, 2018-10-18:
I would like to make a temperature sensor on a battery, but my question is how to do it so that I do not get the not responding
when eps is in deep sleep

Kristian8606, 2018-10-18:
and what is the way to do it?

renssies, 2018-10-19:
@Kristian8606 there is not really a way to do it with sleeping according to the (un)official non-commercial spec even Bluetooth devices can't sleep. 

Since the ESP8266 and ESP32 use a lot of battery while not in sleep mode it's not really suitable for a battery powered sensor. The only way to do it would be to create one that uses mqtt and homebridge. 

There was also a library for the nRF52832, but it's no longer maintained. The nRF52832 is a low power Bluetooth microcontroller

Kristian8606, 2018-10-19:
Thanks for the answer I suspected it would not happen without a bridge.

peros550, 2018-10-19:
Perhaps the problem could be solved by using two devices instead of one. The first one would run on battery and would act as a remote sensor. As often as you wish, the device would wake up, read the temp sensor and transmit the value to the second device. 

The second device would run on wired power and would run homekit firmware.

The connection between the two devices could be something that does not require a constant connection, thus it would be power efficient. (using i.e. RF 466Mhz or X-bee or nrf24L01 modules).  

The second device would receive measurements wirelessly. While there is no new mesaurement, the second device would publish the last known measurement to the homekit. 

Someting like that, I hope I made my point.

NorthernMan54, 2018-10-19:
Just to change the discussion but

1 - Cheap battery powered temperature sensor 433 MHz based ( acurite makes these )

2 - Raspberry PI and homebridge 

3 - rtl_433 and a rtl-Sdr USB dongle to receive the signals.

I’m a big proponent of the ESP8266 for things, but for battery powered temperature this is likely to be a lot better.

Kristian8606, 2018-10-19:
Thank you! I currently have a battery thermometer that works through Domiticz and Homebridge. But I thought how to implement a thermometerу like Eve Degree but with esp obviously will not happen.

maccoylton, 2018-10-29:
Based on my research for battery powered sensors based on my research only bluetooth will give a reasonable battery life. 

The NRF5 based series bluetooth modules seems promising 

There is no up to date library that I know of currently for supporting that natively. 

This one I have got to work but is based on an early version of the protocol https://github.com/aanon4/HomeKit, got it to compile not been able to add new servers due to the old protocol nature

This https://github.com/adafruit/Adafruit_nRF52_Arduino/librarries/BLEHomekit contains a work in process which at the moment doesn't do much more than the ability to pair

Kristian8606, 2018-10-29:
the second link does not open

Kristian8606, 2018-11-08:
Is it possible for the temperature sensor to receive data via HTTP and if possible would you give me a sample code?

maximkulkin, 2019-01-27:
@Kristian8606 yes, this is probably possible, but I won't spend time implementing it.

closed 2019-01-27 by maximkulkin

peros550, 2020-06-15:
@Kristian8606 have you further pursuit this topic? If yes, do you have any sample code to start with?

Kristian8606, 2020-06-15:
@peros550 yes I have sample code that works but is not complete.  I added a LoRa module to esp8266 which I configured as a bridge with multiple sensors.  I started the Arduino with a LoRa battery module.  But I stopped there.  I came up with the idea to add a c2530 zigbee coordinator.