maximkulkin/esp-homekit-demo #138: Wifi not configured from wifi.h file

MaxDiOrio, 2018-10-27:
First - this is absolutely awesome and works amazingly well.  Thank you.

I edited wifi.h for my network, complied and flashed the sonoff_basic example.  After flashing, it just repeated scandone over and over again even after a reboot.  I noticed that it brought up a wifi network, and I was able to connect to it and join my home wifi and it worked fine after that and I was able to add it to HomeKit without any issues.

Just thought it strange the wifi.h didn't work as intended.  The only "strange" thing about my network is that it has a hyphen in it, but I doubt that would be a problem.

Thanks for all the work you've put into this.  It really opens up a lot of opportunities to automate things with this amazingly cheap Sonoff modules.

dachoeks3, 2018-10-31:
That example uses wifi_config.h, so you have to connect to the ESP's wifi to configure it. You can change the wifi name on line 194 (wifi_config_init("sonoff-switch", NULL, on_wifi_ready);) where default wifi name is sonoff-switch.

maximkulkin, 2019-01-27:
There are two kinds of examples: ones that connect to a hardcoded at compile time wifi network and the ones that allow you to configure wifi network at runtime. The former ones use wifi.h and the later don't.

closed 2019-01-27 by maximkulkin