maximkulkin/esp-homekit-demo #140: creation error

Kristian8606, 2018-10-28:
I'm trying to follow these steps, Install esp-open-sdk, build it with make toolchain esptool libhal STANDALONE=n but I get this error

> ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [nconf] Error 1
make: *** [build-lib-kconfig] Error 2
make: *** [build] Error 2
make[1]: *** [_ct-ng] Error 2
make: *** [crosstool-NG/ct-ng] Error 2

maccoylton, 2018-10-29:
might be useful to give a little more background on was platform you are working on and what steps you have taken so far to install

RavenSystem, 2018-10-29:
Are you using a case-sensitive volume to install SDK?

Kristian8606, 2018-10-29:
Yes, I follow the described instructions

RavenSystem, 2018-10-30:
brew tap homebrew/dupes is not necessary. Its content was migrated to the core, so you can install all necessary packages without any problem.

Kristian8606, 2018-10-30:
I still have this mistake

Kristian8606, 2018-10-30:
I solved the problem by installing an older version of Xcode v 9.2 from the apple developer web site

closed 2019-01-27 by maximkulkin