maximkulkin/esp-homekit-demo #141: Add Amazon Alexa to esp8266/32

dachoeks3, 2018-10-29:
Would it be possible to add Alexa controls in conjunction with Apple HomeKit? I have tried to search for examples on RTOS and Amazon Alexa but couldn't find anything.

b0rv3g4r4, 2018-11-06:
Im interested too

NorthernMan54, 2018-11-06:
I’m the developer of the homebridge-alexa skill, and would gladly work on this with someone.

Having esp-HomeKit participate with my skill, should be straight forward as it just uses the HAP API without encryption/pairing.  If you have the homebridge-alexa skill installed on your local network, and your esp-HomeKit device was visible without encryption it should just work.

tomasrocchetti, 2018-12-12:
I'm interested too. I tried homebridge-alexa but it doesn't discover my esp-HomeKit devices

maximkulkin, 2019-09-04:
Closing this as support for Alexa is outside of this project scope

closed 2019-09-04 by maximkulkin