maximkulkin/esp-homekit-demo #143: help OLED

Kristian8606, 2018-11-08:
I'm trying to merge the thermostat code with the ssd1306 display
 help please.

> /*
 * Thermostat example
 *
 * Wiring is as follows (requires ESP-12 as it needs 4 GPIOs):
 *
 * DHT11 (temperature sensor)
 *
 *               -------------
 *              |GND       VCC|    (These go to control pins of relay)
 *              |15         13| --> Heater
 *              |2          12| --> Cooler
 *              |0          14| --> Fan
 *              |5          16|
 * DHT Data <-- |4       CH_PD|
 *              |RXD       ADC|
 *              |TXD      REST|
 *               -------------
 *              |   |-| |-| | |
 *              | __| |_| |_| |
 *               -------------
 *
 */
 #include <espressif/esp_common.h>
#include <queue.h>
#include <timers.h>
#include <string.h>
#include <ssd1306/ssd1306.h>
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <espressif/esp_system.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <etstimer.h>
#include <esplibs/libmain.h>
#include <FreeRTOS.h>
#include <task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"

#include <dht/dht.h>

#define LED_PIN 2
#define TEMPERATURE_SENSOR_PIN 4
#define FAN_PIN 14
#define COOLER_PIN 12
#define HEATER_PIN 13
#define TEMPERATURE_POLL_PERIOD 10000
#define HEATER_FAN_DELAY 30000
#define COOLER_FAN_DELAY 0

#define LOAD_ICON_X 54
#define LOAD_ICON_Y 42
#define LOAD_ICON_SIZE 20

#define CIRCLE_COUNT_ICON_X 100
#define CIRCLE_COUNT_ICON_Y 52

/* Remove this line if your display connected by SPI */
#define I2C_CONNECTION

#ifdef I2C_CONNECTION
    #include <i2c/i2c.h>
#endif
#include "fonts/fonts.h"
[... 259 lines cut ...]
            &current_humidity,
            NULL
        }),
        NULL
    }),
    NULL
};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void user_init(void) {
    uart_set_baud(0, 115200);
	
	screen_init();
    
    wifi_init();
    thermostat_init();
    
    
    
    homekit_server_init(&config);
}

maccoylton, 2018-11-09:
Checkout my thermostat example https://github.com/maccoylton/homekit_wifi_thermostat/blob/master/src/wifi_thermostat.c

maccoylton, 2018-11-18:
try incrementing the stack size to 512k 

instead of 

xTaskCreate(ssd1306_task, "ssd1306_task", 256, NULL, 2, NULL);

try 

xTaskCreate(ssd1306_task, "ssd1306_task", **512**, NULL, 2, NULL);

closed 2019-01-27 by maximkulkin

maximkulkin, 2022-11-15:
@xrust83 Can you show the complete source code

maximkulkin, 2022-11-29:
@xrust83 The code seem fine. Can you describe the issue in more detail: you're saying that you press the "up" or "down" button, the screen lights up, but it shows the temperature that was set before? Does the temperature change in Home app?
Then, you're saying that if you press the "up" or "down" button again, then the screen is updated with a new temperature?

maximkulkin, 2022-11-29:
@xrust83 Gotcha. In this case you need to modify [up_button_callback()](https://github.com/xrust83/homekit_wifi_thermostat/blob/master/src/wifi_thermostat.c#L485) and [down_button_callback()](https://github.com/xrust83/homekit_wifi_thermostat/blob/master/src/wifi_thermostat.c#L498) to check if screen is on. If not, just turn on screen and exit. Otherwise, modify temperature AND extend screen off delay:

```c
void up_button_callback(uint8_t gpio, void* args, const uint8_t param) {
    printf("Button UP single press\n");

    bool modify_temperature = screen_on;
    switch_screen_on(SCREEN_DELAY); /* ensure the screen is on and extend its off timer */
    
    if (!modify_temperature) return;

    if (target_temperature.value.float_value <= (target_temperature.max_value[0] - 0.5))
    {
        target_temperature.value.float_value += 0.5;
        sdk_os_timer_arm (&save_timer, SAVE_DELAY, 0 );
        homekit_characteristic_notify(&target_temperature, target_temperature.value);
    }
}
```
(same applies to down_button_callback()).