maximkulkin/esp-homekit-demo #147: /projekt/esp-open-sdk/examples/led$ make Makefile:13: /common.mk: not found why??

Repair-Lab, 2018-11-24:
I've been on the LED for 1 week now. to create but every time I make a make it does not find it ....
ha@ha-VirtualBox:~/projekt/esp-open-sdk/examples/led$ make
Makefile:13: /common.mk: Datei oder Verzeichnis nicht gefunden<<<<<<<< mean not find<<<<
make: *** Keine Regel, um „/common.mk“ zu erstellen.  Schluss.
with blinky file works great ........ projekt/esp-open-sdk/examples/blinky .
what i make wrong ???
you can help me for solved ??
specal thanks to maximkulkin
 for this preticular programm :-*

Repair-Lab, 2018-11-24:
I found that and I hope that one of you explains it a little better because I am very far and I do not want to give up the middle trough:)

@bwicks What was the issue? 
I see the first issue was due to unset SDK_PATH environment variable.  The second one is due to ESP-SDK toolchain was not in PATH. Is that it?
--

peros550, 2018-12-23:
Make sure your user account has proper permissions to your project's folder and SDK_PATH is set as environmental variable to show path towards the esp-open-rtos folder.

closed 2019-01-27 by maximkulkin