maximkulkin/esp-homekit-demo #153: I don't see my esp32 in HomeKit

velocityzen, 2018-12-11:
Thanks for the demo.

I flashed my ESP32-PICO-KIT V4.1 everything is ok and seems to work, it is connected to wi-fi, I can see device in the network, but I can't add it to Home app on my iPhone. It did not appear there.

Is the default code is 111-11-111?

Log:

```
MONITOR
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6224
ho 0 tail 12 room 4
load:0x40078000,len:10248
ho 0 tail 12 room 4
load:0x40080400,len:6616
entry 0x40080764
I (32) boot: ESP-IDF v3.3-beta1-4-g23b6d40c5 2nd stage bootloader
I (33) boot: compile time 19:35:00
I (33) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : DIO
I (47) boot: SPI Flash Size : 4MB
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x24a3c (150076) map
I (150) esp_image: segment 1: paddr=0x00034a64 vaddr=0x3ff80000 size=0x00000 (     0) load
I (150) esp_image: segment 2: paddr=0x00034a6c vaddr=0x3ff80000 size=0x00000 (     0) load
I (156) esp_image: segment 3: paddr=0x00034a74 vaddr=0x3ffbdb60 size=0x0318c ( 12684) load
I (170) esp_image: segment 4: paddr=0x00037c08 vaddr=0x3ffc0cec size=0x00000 (     0) load
I (174) esp_image: segment 5: paddr=0x00037c10 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/velocityzen/Projects/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (184) esp_image: segment 6: paddr=0x00038018 vaddr=0x40080400 size=0x07ff8 ( 32760) load
I (206) esp_image: segment 7: paddr=0x00040018 vaddr=0x400d0018 size=0x8af60 (569184) map
0x400d0018: _stext at ??:?

I (405) esp_image: segment 8: paddr=0x000caf80 vaddr=0x400883f8 size=0x08e6c ( 36460) load
0x400883f8: rcGetSched at ??:?

I (421) esp_image: segment 9: paddr=0x000d3df4 vaddr=0x400c0000 size=0x00000 (     0) load
I (421) esp_image: segment 10: paddr=0x000d3dfc vaddr=0x50000000 size=0x00000 (     0) load
I (428) esp_image: segment 11: paddr=0x000d3e04 vaddr=0x50000000 size=0x00000 (     0) load
I (447) boot: Loaded app from partition at offset 0x10000
I (447) boot: Disabling RNG early entropy source...
I (448) cpu_start: Pro cpu up.
I (452) cpu_start: Application information:
I (457) cpu_start: Project name:     led
I (461) cpu_start: App version:      e08781d-dirty
I (467) cpu_start: Compile time:     19:35:03
I (472) cpu_start: Compile date:     Dec 10 2018
I (477) cpu_start: ESP-IDF:          v3.3-beta1-4-g23b6d40c5
I (484) cpu_start: Starting app cpu, entry point is 0x40081048
0x40081048: call_start_cpu1 at /Users/velocityzen/Projects/esp/esp-idf/components/esp32/cpu_start.c:261

I (0) cpu_start: App cpu up.
I (494) heap_init: Initializing. RAM available for dynamic allocation:
I (501) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (507) heap_init: At 3FFC73D0 len 00018C30 (99 KiB): DRAM
I (513) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (520) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (526) heap_init: At 40091264 len 0000ED9C (59 KiB): IRAM
I (532) cpu_start: Pro cpu start user code
I (215) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (383) wifi: wifi driver task: 3ffb7948, prio:23, stack:3584, core=0
I (383) wifi: wifi firmware version: 328353e
I (383) wifi: config NVS flash: enabled
I (393) wifi: config nano formating: disabled
I (393) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (403) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (443) wifi: Init dynamic tx buffer num: 32
I (443) wifi: Init data frame dynamic rx buffer num: 32
I (443) wifi: Init management frame dynamic rx buffer num: 32
I (443) wifi: Init static rx buffer size: 1600
I (453) wifi: Init static rx buffer num: 10
I (453) wifi: Init dynamic rx buffer num: 32
W (453) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (633) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 2
I (643) wifi: mode : sta (d8:a0:1d:42:8d:98)
STA start
I (763) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1753) wifi: state: init -> auth (b0)
I (1753) wifi: state: auth -> init (8a0)
I (1753) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
STA disconnected
STA disconnected
I (4293) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (4293) wifi: state: init -> auth (b0)
I (4293) wifi: state: auth -> assoc (0)
I (4303) wifi: state: assoc -> run (10)
I (4333) wifi: connected with test2g, channel 1, bssid = 18:9c:27:cf:06:73
I (4333) wifi: pm start, type: 1

I (6383) event: sta ip: 192.168.1.11, mask: 255.255.255.0, gw: 192.168.1.1
WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x200000
>>> HomeKit: Generated new accessory ID: 97:E7:26:92:A9:FF
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
```

dombergy, 2018-12-15:
Hi,
I am having the same problem, everything in the setup seems to have worked 
(`connected with **myssid**, channel 1
dhcp client start...
ip:192.168.0.113,mask:255.255.255.0,gw:192.168.0.1`)
but can't see the esp8266 in my home app devices, there's just nothing showing up.
I too am looking for help. The default code seems to indeed be 111-11-111 from the code:
`homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};`
Will try to flash other examples and will let you know I get lucky (currently flashed led). Please do the same if you find something.
Cheers.

velocityzen, 2018-12-17:
@velocityzen I think the problem that device doesn't announce himself on mDNS. It also should announce itself for pairing. There is probably some trigger, I'll look into code later today.

velocityzen, 2018-12-21:
I use the bonjour browser to see what is going on mDNS. Device doesn't appear at all. I tried mDNS example from esp-idf and it works perfectly.

dombergy, 2019-01-04:
Hey @velocityzen , thanks for the follow up. Are you saying you can now find the device in the Home app? If so, could you point me to what you did? I don't quite understand how the mDNS protocol works. Also, are you able to control the esp through the app (toggle on/off an output)?
Cheers

maximkulkin, 2019-01-27:
Try building it with latest ESP-IDF (>=v3.2). Prior versions had a bug with mDNS implementation.

Will close it for now, if that does not work, feel free to reopen.

closed 2019-01-27 by maximkulkin