maximkulkin/esp-homekit-demo #157: Sonoff 4CH or 4CH Pro

tgreer, 2018-12-30:
Is there any possibility of a demo bit of code being done for the 4CH sonoff? Tasmota has one  but I don't want to rely on a middleman device.

Thanks

RavenSystem, 2019-01-05:
https://github.com/RavenSystem/esp-homekit-devices

maximkulkin, 2019-01-27:
Mostly the challenge of supporting 4CH and 4CH Pro is knowing which pin is used to control particular  relay for which you can easily find information in internet (or using multimeter and tracing which which esp8266 pin is connected to relay's control pin).

closed 2019-01-27 by maximkulkin

KevinMiddleton, 2019-09-24:
I am hoping to figure this out as well but I am not a C coder.  I tested the RavenSystem but it was not exactly what I was looking for.

The Sonoff 4CH R2 pins are as follows:

Switch   Button       Relay/LED
L1          GPIO0        GPIO12
L2          GPIO9        GPIO5
L3          GPIO10      GPIO4
L4          GPIO14      GPIO15

WiFi LED  GPIO13

Any help in getting the main.c file for this would be appreciated.