maximkulkin/esp-homekit-demo #160: Is it possible to use Arduino SDK?

glararan, 2019-01-10:
Hi,

Last night I googled about homekit on esp32 using arduino sdk. I have to say I didnt find much. This project seems solid and to work. I didnt find out if it is possible to run it with arduino sdk. My program uses few arduino libraries which are dependent on arduino sdk. I would like to add homekit feature without homebridge.

Is it possible? Is there any example/tutorial?

Thanks

maximkulkin, 2019-01-26:
No, this project relies on esp-open-rtos SDK and is not compatible with Arduino SDK.

closed 2019-01-26 by maximkulkin

Rjayone, 2019-05-04:
@maximkulkin Does it possible to make integration with arduino ide without open rtos? I thought, you only need to confirm the HAProtocol?

maximkulkin, 2019-05-04:
I do not care. Supporting other platforms is possible in Arduino IDE, but apart to having a button to upload compiled firmware, Arduino is a pretty shitty IDE targeted mostly to write 20 line sketches. So nothing I would miss about it.

mriksman, 2020-01-08:
@Rjayone I managed to get it running in Arduino IDE

Rjayone, 2020-01-08:
> @Rjayone I managed to get it running in Arduino IDE

Could you describe the process or give any information how i to do this?

mriksman, 2020-01-08:
> 
> 
> > @Rjayone I managed to get it running in Arduino IDE
> 
> Could you describe the process or give any information how i to do this?

I created it from scratch. But it might be easier to grab it from here https://github.com/Yurik72/ESPHap