maximkulkin/esp-homekit-demo #161: Push Notifications

nonameplum, 2019-01-18:
Hey,
I checked sketchy in the HAP Specification but I couldn't find it.
Is there a way with this fantastic library to allow our custom ESP devices to be push notifications enabled for the HomeKit?
For example, we could send push notification on the led state change or whatever any other reason?
Currently when we add our custom ESP device to the HomeKit we don't have this option available:
[image]
I assume that device sends push notification directly by the HAP but maybe I'm mistaken and push notification is sent by the device as any other push notification by APNS server.

renssies, 2019-01-18:
Unfortunately, this is not possible. The push notifications feature of the Home app works through the Home Hub (either an Apple TV, HomePod or iPad). The Home hub starts observing the events coming from an accessory and sends out a notification if wanted by the user. 

Which accessories and events can have push notifications is determined by Apple in the Home app and is outside the HomeKit spec.

This sadly has nothing to do with the accessory itself

nonameplum, 2019-01-18:
@renssies  Ahh now everything is clear. Thanks for the great explanation. Do you have any source to read about this more? I'm wondering if Home Hub is sending a push notification via APNS server or they have some special private API that they send a push notification to the listening devices from Home Hub or some sort of information that Home app is handling and sending a local notification.

NorthernMan54, 2019-01-18:
@nonameplum have you looked at using automations to trigger something when a device is controlled?

If your goal is trigger a iOS notification, have you looked at homebridge for a fake sensor that creates iOS notifications ?  In homebridge, it wouldn’t be that hard for a developer to create a fake motion sensor triggered by a fake switch.  You could then have an automation trigger the fake switch when your led is on.

nonameplum, 2019-01-18:
@NorthernMan54  Yup, that should work.
I think also about a different solution. We could use something like https://pushed.co/for-developers with https://itunes.apple.com/us/app/get-pushed/id804777699?mt=8&at=&ct=&ign-mpt=uo%3D6.
I think we could directly send push notification from our accessory device. This wouldn't require any additional server or Home Hub. Of course notification won't be handled by the Home app but still, the most important fact about some state change will be noted.

renssies, 2019-01-19:
@renssies I mostly got that information from looking around in the available metadata files that are in a decoded IPSW. Also, thttps://twitter.com/KhaosT and https://twitter.com/Brutella sometimes share details about HomeKit

As for the notifications, I know these notifications are sent from the Apple Server, so they have a special API on the API, probably a private CloudKit API (HomeKit is partly built on CloudKit)

seritos, 2019-02-09:
Hi, 

I have installed this firmware and it has the notifications enabled. I don't have any hub and the work perfectly...

https://www.youtube.com/watch?v=KfzeZMIxj-c

nonameplum, 2019-02-10:
@seritos Is that open source? Isn't that firmware based on @maximkulkin work and again someone is trying to earn money on opensource with license violation?

seritos, 2019-02-10:
Hi @nonameplum 
I don't think it's an open source. I haven't ever seen the code. I have asked him to send me the bin file and he did it after a "donation" for the maintenance of his youtube channel.

nonameplum, 2019-02-10:
If you have a contact you could ask him if his work is based on https://github.com/maximkulkin/esp-homekit then it would be great if he could explain a bit how is that working. We could make a PR with another example of this repo so everyone could learn.

nonameplum, 2019-02-17:
@renssies @NorthernMan54  @seritos  I just checked `door-sensor` example and as I had the first impression that notifications might just work with some type of devices it is actually a truth. I run this example with sonoff SV and reed switch. The Home app will discover this device as read-only sensor device with the "Allow notifications" setting available. After turning this one it just works.

closed 2019-02-26 by maximkulkin