maximkulkin/esp-homekit-demo #162: 433 MHz tx?

DominikTo, 2019-01-19:
Did anyone build a 433 MHz TX project on this? E.g. to remote control cheap 433 MHz outlets via HomeKit?

maccoylton, 2019-01-19:
There is a 433MHz library that has been submitted as an extra for esp-open-rtos, it should be fairly easy to use this to build a bridge style device. It can be found here:- https://github.com/Felix5721/rf433mhz

NorthernMan54, 2019-01-19:
I did this via homebridge 

https://github.com/NorthernMan54/homebridge-HTTP-IRBlaster

I found the challenge was around managing the transmitter codes, I have several devices and just use one esp8266/tx module to control them.  While in homebridge I have about a dozen devices configured

closed 2019-02-26 by maximkulkin

lukasa1993, 2019-10-23:
why this was closed would be cool to have such option

maximkulkin, 2019-10-23:
It might indeed be cool, but.. I do not plan to implement this and I'm not aware if anybody else plans to, so what's the point of having it hanging in list of open issues?

lukasa1993, 2019-10-23:
is implementing this harder then other samples?

maximkulkin, 2019-10-23:
I do not know and why does it matter?