maximkulkin/esp-homekit-demo #166: Lock Management

eryk4pdh, 2019-02-03:
Is there a chance someone can help me with an example for Lock Management device?

Thank you

maximkulkin, 2019-02-03:
Start by asking specific questions

eryk4pdh, 2019-02-04:
I would like to create an ESP home kit enabled lock that... The ESP would be connected to an electric door lock controlled via a relay...  [(](https://www.amazon.co.uk/Mirocle-ESP8266-7-30V-Network-Module/dp/B01L5H0PZG)

I guess this would have to use the 'Lock Management' or Lock Mechanism home kit service....

I don't know where to start with this can someone point me to some docs about this?

maximkulkin, 2019-02-04:
Well, that ESP relay is fine, but I hope you understand that you also need an actual lock which has some kind of actuator/magnet lock which you will control with a relay.

So, according to HAP spec, your lock needs to have two services - Lock Mechanism (primary service) and Lock Management. Lock Mechanism is pretty straight forward: you have current lock state and desired lock state. If your lock mechanism does not provide you feedback on current state, just return desired state as current state (assume everything works).

Lock Management is there to allow vendors to implement custom control mechanisms (like managing users who can lock/unlock door), but you probably do not need that. So, Lock Control Point characteristic can be left empty (it's write only characteristic, just do not do anything there). Version characteristic is just a string.

eryk4pdh, 2019-02-04:
Is this based on https://www.espressif.com/en/products/software/esp-homekit-sdk ??

maximkulkin, 2019-02-04:
No, it's not. Thing is that to produce and sell HomeKit compatible accessories, your company need to be certified for that (https://developer.apple.com/homekit/, `If you’re interested in developing or manufacturing a HomeKit accessory that will be distributed or sold, your company must enroll in the MFi Program.`) Espressif have their implementation of HomeKit framework, but it will give you it only if you have MFi certification.

This project is a non-commercial implementation of HAP protocol, not meant for commercial use.

eryk4pdh, 2019-02-04:
This is for a hobby project ... I want to just find some documentation... how to get started with this...

I think my specialty leans itself more on the hardware engineering, so I a not proficient at coding so I always tend to rely on some documentation to get me going.... Thanks for help

eryk4pdh, 2019-02-04:
Thanks i will try to do something

closed 2019-02-04 by eryk4pdh

hadaroren, 2019-10-03:
hi,
I'm also having problem with the lock Management service... did you succeed to implement it in your lock?