maximkulkin/esp-homekit-demo #167: Вопрос

outsiderz, 2019-02-10:
Максим добрый день!
Хочу поинтересоваться у вас возможно ли один модуль ESP представить в виде двух аксессуаров в HomeKit? 1)датчик движения 2) замок. Или для этого нужно 2 модуля ESP? хочу сделать управление домофоном с ip камерой но так как Apple убрали поддержку doorbell, а мне нужно что бы приходили уведомления с камеры.

maximkulkin, 2019-02-10:
Да, для Ваших целей достаточно одного модуля. Более того, поскольку это логически одно устройство с разными функциями, их надо представлять в виде одного аксессуара (HAP Accessory Object, см. раздел 2.5.3 [HAP спецификации](https://developer.apple.com/homekit/specification/)).

И на самом деле Apple не убирали поддержку Doorbell/: https://appleinsider.com/articles/18/08/02/apples-removal-of-the-homekit-doorbell-category-is-no-cause-for-alarm https://www.youtube.com/watch?v=XXyTvPw57r0

outsiderz, 2019-02-15:
Максим а как с вами можно связаться в телеграмм? мой телеграмм @outsiderzEnik

closed 2019-02-16 by maximkulkin