maximkulkin/esp-homekit-demo #168: issue compiling example/led/all

hfosteriv, 2019-02-14:
Hi, 

This is my first post on here so I hope I do it correctly! 

When trying to make the example/led all command I get the error

`CC /Volumes/case-sensitive/esp-open-sdk/esp-open-rtos/open_esplibs/libmain/ets_timer.c`

`make: xtensa-lx106-elf-gcc: No such file or directory`

`make: *** [build/open_esplibs_libmain//ets_timer.o] Error 1`

I have tried this on Ubuntu (all the way updated, fresh install on virtual machine) and in macOS but keep getting the error. 

Also on both machinesWhen ring to run the make STANDALONE command it always ends with:

`checking how to run the C preprocessor... gcc -E`

`checking for ranlib... ranlib`

`checking for gobjcopy... no`

`checking for objcopy... no`

`configure: error: missing required tool: gobjcopy objcopy`

`make[1]: *** [_ct-ng] Error 1`

`make: *** [crosstool-NG/ct-ng] Error 2`

Don't know if these are related but any help would be appreciated. 

I did clone the repositories and set the PATH and SDK_PATH correctly I believe, its just missing the resource(?)

Also, I have looked into others having the same problem with the gobjcopy/objcopy and they noted it should not necessarily be an issue for native work, however I am not sure if that is the case or not.

hfosteriv, 2019-02-15:
I went through and redid everything the same again, but now I am getting a different error. 

[INFO ]  Installing cross-gdb
[ERROR]    collect2: error: ld returned 1 exit status
[ERROR]    make[5]: *** [gdb] Error 1
[ERROR]    make[4]: *** [all-gdb] Error 2
[ERROR]    make[3]: *** [all] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing cross-gdb'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@120]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 42:47.87)
[42:48] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/home/parallels/esp-homekit-demo/esp-open-sdk/crosstool-NG'
../Makefile:135: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/parallels/esp-homekit-demo/esp-open-sdk/crosstool-NG'
Makefile:130: recipe for target 'crosstool-NG/.built' failed
make: *** [crosstool-NG/.built] Error 2

not sure about this one either, I have plenty of free disk space in both the .dmg file (11gb) on my main system (macOS) but also enough space available for the Ubuntu virtual machine being ran through parallels.

Any help would be appreciated.

hfosteriv, 2019-02-16:
@maximkulkin

maximkulkin, 2019-02-16:
This is the problem with esp-open-rtos, not with esp-homekit. You need to report to that project.

closed 2019-02-16 by maximkulkin

hfosteriv, 2019-02-16:
Its always what I obvious, isn't it. 

Thanks!