maximkulkin/esp-homekit-demo #169: Control two led strip with same ESP8266

MrGh0ST, 2019-02-14:
Hello,
I would like to control two led strip with the same ESP8266, using 2 GPIO, but I do not know how to do it.

But especially to control them independently.

Do you have an idea ?

MrGh0ST, 2019-02-14:
`/*
* This is an example of an rgb ws2812_i2s led strip
*
* NOTE:
*    1) the ws2812_i2s library uses hardware I2S so output pin is GPIO3 and cannot be changed.
*    2) on some ESP8266 such as the Wemos D1 mini, GPIO3 is the same pin used for serial comms.
* 
* Debugging printf statements are disabled below because of note (2) - you can uncomment
* them if your hardware supports serial comms that do not conflict with I2S on GPIO3.
*
* Contributed March 2018 by https://github.com/Dave1001
*/
#include <stdio.h>
#include <stdlib.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>
#include <math.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"
#include "ws2812_i2s/ws2812_i2s.h"

#define LED_ON 0                // this is the value to write to GPIO for led on (0 = GPIO low)
#define LED2_ON 0                // this is the value to write to GPIO for led on (0 = GPIO low)
#define LED_INBUILT_GPIO 2      // this is the onboard LED used to show on/off only
#define LED_COUNT 16            // this is the number of WS2812B leds on the strip
#define LED_COUNT_2 16            // this is the number of WS2812B leds on the strip
#define LED_RGB_SCALE 255       // this is the scaling factor used for color conversion
#define GPIO_OUTPUT_2 4

// Global variables
float led_hue = 0;              // hue is scaled 0 to 360
float led_saturation = 59;      // saturation is scaled 0 to 100
float led_brightness = 100;     // brightness is scaled 0 to 100
bool led_on = false;            // on is boolean on or off
bool led2_on = false;            // on is boolean on or off

ws2812_pixel_t pixels[LED_COUNT];
ws2812_pixel_t pixels2[LED_COUNT_2];

//http://blog.saikoled.com/post/44677718712/how-to-convert-from-hsi-to-rgb-white
static void hsi2rgb(float h, float s, float i, ws2812_pixel_t* rgb) {
    int r, g, b;

    while (h < 0) { h += 360.0F; };     // cycle h around to 0-360 degrees
    while (h >= 360) { h -= 360.0F; };
    h = 3.14159F*h / 180.0F;            // convert to radians.
    s /= 100.0F;                        // from percentage to ratio
    i /= 100.0F;                        // from percentage to ratio
    s = s > 0 ? (s < 1 ? s : 1) : 0;    // clamp s and i to interval [0,1]
    i = i > 0 ? (i < 1 ? i : 1) : 0;    // clamp s and i to interval [0,1]
    i = i * sqrt(i);                    // shape intensity to have finer granularity near 0

    if (h < 2.09439) {
        r = LED_RGB_SCALE * i / 3 * (1 + s * cos(h) / cos(1.047196667 - h));
        g = LED_RGB_SCALE * i / 3 * (1 + s * (1 - cos(h) / cos(1.047196667 - h)));
        b = LED_RGB_SCALE * i / 3 * (1 - s);
    }
    else if (h < 4.188787) {
        h = h - 2.09439;
        g = LED_RGB_SCALE * i / 3 * (1 + s * cos(h) / cos(1.047196667 - h));
        b = LED_RGB_SCALE * i / 3 * (1 + s * (1 - cos(h) / cos(1.047196667 - h)));
        r = LED_RGB_SCALE * i / 3 * (1 - s);
    }
    else {
[... 318 lines cut ...]
homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"
};

void user_init(void) {
    // uart_set_baud(0, 115200);

    // This example shows how to use same firmware for multiple similar accessories
    // without name conflicts. It uses the last 3 bytes of accessory's MAC address as
    // accessory name suffix.
    uint8_t macaddr[6];
    sdk_wifi_get_macaddr(STATION_IF, macaddr);
    int name_len = snprintf(NULL, 0, "Sample LED Strip-%02X%02X%02X", macaddr[3], macaddr[4], macaddr[5]);
    char *name_value = malloc(name_len + 1);
    snprintf(name_value, name_len + 1, "Sample LED Strip-%02X%02X%02X", macaddr[3], macaddr[4], macaddr[5]);
    name.value = HOMEKIT_STRING(name_value);

    wifi_init();
    led_init();
    homekit_server_init(&config);
}`

I'm test this, but is not functionnal :/
If you have same idea ?

maximkulkin, 2019-02-14:
```c
// initialise the LED strip
ws2812_i2s_init(LED_COUNT, PIXEL_RGB);
ws2812_i2s_init(LED_COUNT_2, PIXEL_RGB);
```
So, the problem here is that you're using example that uses ws2812_i2s library which uses I2S interface to program LED strip, but ESP8266 has only one I2S output interface (on pin 3).

Instead you need to use [ws2812](https://github.com/SuperHouse/esp-open-rtos/tree/master/extras/ws2812) library (notice: no "_i2s" suffix) which uses plain bit banging to send data. It is known that it can be imprecise under heavy load, but in general it should be fine.

So, to change your implementation you need to get rid of `ws2812_i2s_init()` calls and replace `ws2812_i2s_update()` calls with something like
```c
ws2812_seq_start();
ws2812_set_many(LED1_GPIO, pixels, LED1_LED_COUNT);
ws2812_seq_end();
```
Don't forget to replace `extras/ws2812_i2s` with `extras/ws2812` in Makefile and include proper header file.

MrGh0ST, 2019-02-14:
I'm not sure I understand the whole process.

I have to use the ws2812 library alone to control both led strip?

maximkulkin, 2019-02-14:
Yes, you need to use ws2812 library instead of ws2812_i2s library since the later one can only control one particular pin and you need to control 2 different pins.

MrGh0ST, 2019-02-14:
Yes, i understand. I try, thanks ;)

MrGh0ST, 2019-02-15:
`void led_init() {
    // initialise the onboard led as a secondary indicator (handy for testing)
    gpio_enable(LED_INBUILT_GPIO, GPIO_OUTPUT);
    gpio_enable(LED_INBUILT_GPIO, GPIO_OUTPUT_2);

    ws2812_seq_start();
    ws2812_set_many(GPIO_OUTPUT, pixels, LED_COUNT);
    ws2812_set_many(GPIO_OUTPUT_2, pixels2, LED_COUNT_2);
    ws2812_seq_end();

    // set the initial state
    led_string_set();
}`

Is not working, i change the library for ws2812 in Makefile and in my C file.

You have an idea ?

maximkulkin, 2019-02-15:
This is completely messed up. Read my suggestions carefully.

MrGh0ST, 2019-02-18:
I read your suggestions, but i understand what i miss :/
You not have a example ?

Thx for that

maximkulkin, 2019-02-18:
No, there is no example.

Examples are mostly provided by community, people implement stuff for their particular needs. To make a good example (that works out of the box) requires testing in hardware which requires assembling actual circuits which requires a) parts; b) time.

The project mostly targets developers who either know or can figure out what they are doing. It is not meant to be a firmware catalog for any occasion.

closed 2019-03-11 by maximkulkin