maximkulkin/esp-homekit-demo #170: Devices don't work remotely through Home Hub (Apple TV)

wasp100, 2019-02-15:
Hi,
I noticed that when trying to control the devices remotely (I have an Apple TV which acts as the Home Hub), the devices show up as "No Response". 

However, when I'm on my local wifi network, the devices work perfectly!

Has anyone else picked this up, and is there a fix for this? I've tried it with the LED code as well as the Sonoff code.

Thanks!

wasp100, 2019-02-15:
@maximkulkin

maximkulkin, 2019-02-15:
Is Apple TV at the same WiFi network as devices?

I have Apple TV and it works great: e.g. I have an automation that turns on living room lights at 8pm.

wasp100, 2019-02-15:
Yes, it’s on the same network. 
All my other HomeKit devices work perfectly with the Apple TV. 
I’ve restarted the Apple TV as well, but no luck. 
Anything else I should try?

wasp100, 2019-02-15:
What I noticed is when I’ve disconnected the power from the device and reconnected (to simulate a power outage), the device would show as No Response over cellular, yet connect when on local Wifi. 
Can you perhaps try this with your lights? Disconnect the HomeKit device and reconnect and see whether you can control remotely over cellular?

maximkulkin, 2019-02-16:
It does this because iOS controllers are slow to reconnect after connection termination (since Apple TV keeps connections to all accessories) while on iPhone if you just opened the Home app, it reconnects to them. I believe the same would happen with all certified accessories.

wasp100, 2019-02-16:
I tested your hypothesis by switching off my other HomeKit Wi-Fi accessories along with the maximkulkin device. 
The HomeKit Wi-Fi devices returned their states after 2-3 minutes. 
The maximkulkin device is still showing No Respinse over cellular. If I change over to Wi-Fi, it shows the correct state. As soon as I toggle over to cellular, it shows No Response again. :(

wasp100, 2019-02-18:
@maximkulkin 
Have you managed to recreate this with your setup?

peros550, 2019-02-19:
Guys, try to power reset  Apple TV and all Wifi devices (Router, AP, extenders)  when troubleshooting.

closed 2019-09-04 by maximkulkin