maximkulkin/esp-homekit-demo #171: 'No Response' after reboot/power outage esp8266

hfosteriv, 2019-02-18:
So I am finally gaining with developing using this SDK and examples. However An issue I am finding on every single example is that when the power is disrupted to the esp8266, the pairing seems to be disconnected permanently and irreparably afterwards. No I do think a reset button would fix the issue, allowing the user to re-pair the devices together, however that is just annoying and should be able to be resolved from other means. 

I did a little testing and found that it does not matter if the device is paired or not, if the device reboots for any reason at all it won't be broadcasting anything.

When it reboots it breaks the open uart connection for output, but by running the 'monitor' command in the corresponding examples it re-establishes that part connection showing not output. I have even went as far as to put a `printf("in function x\n")` in every function and nothing is printing at all. 

Now I did see that some people are experiencing similar issues sporadically upon reboots, however this is every single boot in my case, which lead me to open another discussion about it since it seems to be little different, and getting lost in the issue posts. 

I am not getting any logs after the reboot myself, others may be but I am not. 

Anyone else having this issue? Any luck finding resolutions to it?

hfosteriv, 2019-02-19:
I am starting to wonder if this is hardware related. 

Some of the post on here are saying that it is hardly/if ever and issue for the ESP-12F, and I have a spark fun esp8266 thing dev that I tried and an ESP-01 hat I tried and both failed after a reboot. Personally have not been able to test a ESP-12F but I have one in the mail, should be here by Wednesday.

I went ahead and inspected the connections of the ESP8266EX on the data sheets of the 3, paying special attention to the spark fun and the ESP-12F. The only 'big' difference I see is that on the ESP-12F pin 32 is connected to VCC through a 12k resistor as well as GND through a 1uF capacitor. Where as the sparkfun esp8266 thing dev pin 32 ONLY gets the vcc connection through a 10k resistor, no cap to ground (this also holds true for the ESP-01, no cap). Thi is important (I think) because Pin 32 is the "EXT_RSTB - External reset signal (Low voltage level: active)" according to the Espressif data sheet.

Another difference I am seeing is that pin 7, CHIP_EN on the spark fun is supplied with VCC through a 10k resistor, but on the ESP-12F it goes to a jumper. Not having worked with the ESP-12F I am assuming that this is supplied with voltage to use it. 

I have a few higher voltage capacitors at 1uf, but I think that they would be too large for the kind of test needing to be done to see if that is the issue. I am sure I have some salvageable electronics containing what I need, just have to look. 

Anyone else think that a capacitor placed between the esp8266ex pin 32 and GND may solve the issue?

hfosteriv, 2019-02-19:
This is the esp-12f diagram, look at pin 32s connections:

[image]

And here we have the spark fun esp8266 thing dev, again look at pin 32. the esp-01 has the same pin 32 circuitry as the spark fun does: 

[image]

Both the spark fun esp8266 thing dev and the esp-01 have the issue with no response after restarting AND are also lacking that decoupling capacitor on the reset pin. 

For those that don't know, the application of the capacitor is that when the system is given voltage the capacitor will 'fill' to match the voltage, but when the system voltage changes(ie power failure, reboot) the capacitor will try to equalize its held voltage with the systems input voltage by discharging over X amount of time. Which I think is where we are getting the issue with restarting, because the system on the chips without the capacitor is getting hung up and never resetting.

hfosteriv, 2019-02-19:
@maximkulkin from your professional point of view, would this make sense?

maximkulkin, 2019-02-26:
I think that capacitor is meant to debounce reset button press (after all, this pin is for resetting chip) and spark fun esp8266 board uses DTR serial signal to do reset, so no debouncing is needed.

From your description it feels like chip either does not start at all or starts in flashing mode. Here is [Spark Fun ESP8266 diagram](https://cdn.sparkfun.com/datasheets/Wireless/WiFi/ESP8266ThingDevV1.pdf), notice `GPIO0: On bootup will run program if high and bootloader if low
Tied to DTR to run bootloader when reset`. Looks like you need to pull it high to run program on reset. GPIO pin is 4th pin from top on right side on that diagram.

This is kind of unnatural thing for ESP8266 boards, Spark Fun try to be creative. I really recommend using [NodeMCU boards](https://www.amazon.com/HiLetgo-Internet-Development-Wireless-Micropython/dp/B010N1SPRK/ref=sr_1_1_sspa), they have less such surprises.

hfosteriv, 2019-02-26:
That’s an interesting note about goio0 that I overlooked, or it’s not on the schematic anyways. I will look into that more in a little while, But does the ESP-12F have/use in implementation that pin high for running? I did not see that in the schematic either is why I ask, but those chips are for more practical application and a pull-up resistor could easily be applied in an broader system setup.

hfosteriv, 2019-02-26:
Glad to see this discussion is getting the community to a diagnosis and hopefully resolution!

wasp100, 2019-03-19:
@maximkulkin 
Hi. I have the same problem. I have flashed the sonoff basic and all works well until there is a power outage. The device now shows up as No Response. 
Has there been any resolution to this?

smilespray, 2019-03-25:
Also getting this on all my Sonoff Basic devices.

kunjal83, 2019-03-27:
I have observed this behaviour on ESP8266 when I did load a bad compiled code. But, I am not experiencing any issues when I try to mimic the power failure by manually switching off the power and abruptly starting it on again after few mins / hours / even next day. 

Have you tried re-flashing the binary?

hfosteriv, 2019-03-27:
@kunjal83 I would ask what module you are using for the esp8266 as for the purpose of what we are speaking of, each one while different is based on the esp8266 SoC but from module to module there are slight hardware differences that may be related to why some modules (the spark fun thing/thing dev, esp01) have issues rebooting but others (esp12f) retart fine. 

to answer your question yes, I have refreshed the binary several times. I personally feel as though on the spark fun thing dev it has to do with the auto reset configuration because during reset it is automatically pt into the programable mode.

maximkulkin, 2019-04-22:
Was the issue resolved?

closed 2019-04-27 by maximkulkin