maximkulkin/esp-homekit-demo #172: Build for ESP32: Fatal error: etstimer.h: No such file

Christian1313, 2019-02-21:
I followed the instructions up to the point 5. Build example.

I deleted the component  WS2812FX due to previous build error (see #84 ).

But then I got:

```
... esp-homekit-demo/components/led-status/led_status.c:3:22: fatal error: etstimer.h: No such file or directory
compilation terminated.

make[1]: *** [led_status.o] Error 1

make: *** [component-led-status-build] Error 2
```

maximkulkin, 2019-02-25:
led-status is an example for esp8266. ESP32 examples are under examples/esp32 and right now there is only one example there.

closed 2019-02-25 by maximkulkin

shaurya0406, 2019-03-21:
I am also getting the same build error while compiling esp32/led example.

maximkulkin, 2019-03-21:
ok, this caused by the fact that some of components were esp8266-/esp32-only and build system was dumb enough to help separating them.. So I just pushed a change that splits components into 3 categories.. This should fix problem of component clashes.

Should compile now.

shaurya0406, 2019-03-22:
It compiled successfully.
Thanks