maximkulkin/esp-homekit-demo #175: Sonoff T1 1CH support?

danio0701, 2019-02-24:
Hi,

I have a trouble to find example files to Sonoff T1 1CH light wall swich :(

Any ideas or links where I can find files examples?

thanks a lot for suggestions.

maximkulkin, 2019-02-26:
There is none, but you can contribute one

danio0701, 2019-02-26:
> There is none, but you can contribute one

Ok, Is it possible to download from sonoff touch full configuration with soft or basic? I'm asking because a local store is selling sonoff touch with homekit software and if it possible I can try backup this firmware and upload here :)

danio0701, 2019-02-26:
I founded article how to backup firmware so I will buy from them this sonoff devies, backup firmware and try to write on my sonoff :)

closed 2019-03-25 by danio0701

nhatquang88, 2019-05-19:
> I founded article how to backup firmware so I will buy from them this sonoff devies, backup firmware and try to write on my sonoff :)

would u please share the firmware and how to flash it thank you