maximkulkin/esp-homekit-demo #178: Sonoff power and library HLW8012

oremic, 2019-02-28:
hi.
There are ways to have the HLW8012 library to get the EVE power meter via apple homekit. for th10 and th16
thank you

maximkulkin, 2019-02-28:
Ok, I guess this is a question, not a statement and it's about supporting power measurement on some [Sonoff models](https://www.itead.cc/sonoff-pow.html) ([a good writeup](http://tinkerman.cat/hlw8012-ic-new-sonoff-pow/) by author of HLW8012 library for ESP8266 on ArduinoCore).
Although I couldn't find evidence that HLW8012 is installed on Sonoff TH10/TH16 models.

I checked the library and I think it can be easily ported to esp-open-rtos: you just need to replace digitalWrite() and always use interrupts instead of doing busy waiting counting pulse width in pulseIn() function.

As of Elgato EVE support for exposing that data, here is [reverse engineered list of EVE custom characteristics](https://gist.github.com/gomfunkel/b1a046d729757120907c).

oremic, 2019-02-28:
Hello
There was a question.
It would be really nice to have consumption, for sonoffs.
Eve allows to have a series of information on consumption.
I could test it if you can make a port.
thank you.

seritos, 2019-09-04:
Hi @maximkulkin,

Many devices now have integrated this power meter. In example,  the cost effective Houzetek AW7PL has the HWL8012 and the device is apparently great, as has several GPIOS free to add it aditional functionalities.

Would be possible to integrate it? 

Many thanks!

curla92, 2019-10-08:
Digital write must be replace with gpio_write? Does it right? 

And i must use gpio_set_interrupt instead of pulsein. Right? Let me know thanks

HomeKidd, 2019-11-29:
@maximkulkin I'm also interested  in HWL8012 library for esp-open-rtos 😄

Kristian8606, 2020-04-10:
As Maximkulkin said, this library will be easily modified.  The question is, did anyone manage to create custom features?  @Maximkulkin would you build these features for us users?

HomeKidd, 2020-04-10:
Yes, i have but without the Eve characteristics and HomeKit data type it will not work. HomeKit data type is added to esp-homekit, but its not that easy to work with like an int/float based characteristic like Temp Sensor etc.😄 But i'm still working on it 😄

Kristian8606, 2020-04-10:
Can you share the code? I think maximkulkin added the data type.

HomeKidd, 2020-04-10:
[HLW8012_BL0937_ESP-master.zip](https://github.com/maximkulkin/esp-homekit-demo/files/4460631/HLW8012_BL0937_ESP-master.zip)
As i've said, data type is added because i've requested it but currently on static data is working. But its way more complicated since other types......i'm working on this since december

Kristian8606, 2020-04-10:
In another post you mentioned that you were able to show the story, can you share code?

HomeKidd, 2020-04-10:
> In another post you mentioned that you were able to show the story, can you share code?

I was not able to show any history with esp-homekit, only with HomeKit Accessory Simulator and only one time.  So theres no code currently!

maccoylton, 2020-04-11:
I already did a port.. it's here:-  https://github.com/maccoylton/HLW8012_BL0937_ESP
Example of a device using it it here:- https://github.com/maccoylton/esp-homekit-gosund-up111

Feedback welcome