maximkulkin/esp-homekit-demo #180: Not able to compile

AchimPieters, 2019-03-08:
First off all, great job! thumbs up!

I'm trying now for a few day's to install all necessary perquisites as described in your README.md file.
First I tried to install it on my MAC, but after google myself crazy if found out that crosstool-ngis no longer supported:

> Nov 26, 2018
> macOS is no longer supported
> 
> With me (Alexey) being the only active maintainer, it is no longer possible to debug any issues arising on macOS systems. I do not have a macOS system, and Apple’s license does not allow one to install an image of macOS into a VM to do this kind of development work.
Source: http://crosstool-ng.github.io/

Okay then I took my Raspberry Pi and tried to install all necessary perquisites as described in your README.md. unfortunately without success. then i tried to install it by following this instruction https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup-scratch.html but also unfortunately without success.

My question:

Can or would you please help me? on witch OS do you compile your main.c code to main.bin so they can be flashed to a ESP module? Is there a possibility that you make a bash file so i know everything is installed as it should? Any help would be welcome!

RavenSystem, 2019-03-08:
You can use a virtual machine with any Linux x86_64 distribution.

maximkulkin, 2019-03-08:
I have built a docker container with toolchain and use it on my Mac. Here is [instructions](https://gist.github.com/maximkulkin/bca710444cbbc01a1845687a266f45a4)

hwater, 2019-03-08:
Hello Maxim

I just tried your instructions, but there is an Problem with the last step:

export SDK_PATH=~/projects/esp-open-rtos

Is this Path correct? At what place lies the SDK? I am not used with Docker, but the 
sdk and rtos dockerfile compiled okay.

many greetings from austia
Holger Watermann

> 
> I have built a docker container with toolchain and use it on my Mac. Here is instructions <https://gist.github.com/maximkulkin/bca710444cbbc01a1845687a266f45a4>

howbrook:esp-homekit-demo holger$ make -C examples/thermostat test
Makefile:14: /Users/holger/projects/esp-open-rtos/common.mk: No such file or directory
make: *** No rule to make target `/Users/holger/projects/esp-open-rtos/common.mk'.  Stop.
howbrook:esp-homekit-demo holger$ pwd
/Users/holger/Documents/Dev_Dock_esp/esp-homekit-demo

maximkulkin, 2019-03-08:
Oh, sure. So this part is not that polished. The problem is that I couldn't get USB access into docker by the time I was writing that (maybe you can do that now), so flashing is actually is done from host machine, not from docker container. Thus, you also need to have esp-open-rtos installed on host machine and SDK_PATH needs to point to that SDK directory. You can get SDK by cloning it from Github:
```
git clone --recursive https://github.com/Superhouse/esp-open-rtos.git
```
and then set SDK_PATH to path where you cloned it. E.g. if you were in home directory and typed command above, you need to `export SDK_PATH=~/esp-open-rtos`

AchimPieters, 2019-03-10:
unfortunately It also didn't work. On witch OS doe you compile you code? maybe I can install a virtual machine as RavenSystem suggested.

maximkulkin, 2019-03-11:
@AchimPieters what exactly did not work?

I'm running on macOS, those instructions do work for me.

hwater, 2019-03-11:
Hello Maxim

Here it runs well with your new rtos-dockerfile. Thanks a lot, this makes it easy om macOS.

Holger

 - typed mobil.
http://donautics.stwst.at/stories-donautics

AchimPieters, 2019-03-11:
Well I'am new to these docker files and can't get them loaded in to docker. I downloaded docker from https://www.docker.com installed it and tried to load your files. Maybe I'm looking to long into this, and need to take a step back to get the overview back. :$

maximkulkin, 2019-03-11:
Well, there is no easy way, you'll have to learn something anyways.

AchimPieters, 2019-03-12:
No of course, what would be the fun in that. :relaxed: I'm gonna try tonight again! :thumbsup:

AchimPieters, 2019-03-12:
YES I did it :bangbang: thank you for your :whale: files. I just had to get my focus back, I was trying to hard for to long. It is really easy and with the little remark from hwater I figured the last part out. Thank you so much and keep up the good work :thumbsup:

closed 2019-03-12 by maximkulkin