maximkulkin/esp-homekit-demo #181: Example "Programmable Switch Event"

GermanSheepDog, 2019-03-08:
Hello Maxim,

I worked fine with esp_homekit and some services, like HOMEKIT_SERVICE_SWITCH,
HOMEKIT_SERVICE_LIGHTBULB, HOMEKIT_SERVICE_CONTACT_SENSOR,
HOMEKIT_SERVICE_TEMPERATURE_SENSOR,...

Now I need to implement HOMEKIT_SERVICE_STATELESS_PROGRAMMABLE_SWITCH with HOMEKIT_CHARACTERISTIC_PROGRAMMABLE_SWITCH_EVENT (with Single Press).

I read the "HAP-Specification-Non-Commercial-Version.pdf", but it did not work.
I like to show only one instance of this service on the accessory.
I only see the Switch Icon and no buttons.

Is it possible that you are creating an example?

maximkulkin, 2019-03-08:
https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/button

closed 2019-03-11 by maximkulkin

GermanSheepDog, 2019-03-23:
Hello Maxim,

I am looking for an example HOMEKIT_SERVICE_STATELESS_PROGRAMMABLE_SWITCH for an ESP32!
Today there is no example and I started with the template of the esp32/led-example to create an esp32/button-example.

If i started the short example I got the same error as in my big program:
```
  Standardraum Button
  Es gibt eine nicht konfigurierte Taste.

  or in english: 

  Standard room button
  There is an unconfigured button.
```
Here are my esp32-example-directory:
[esp32.zip](https://github.com/maximkulkin/esp-homekit-demo/files/2999432/esp32.zip)

Note: Currently I did not implement the time function to read how the long the button was pressed.

GermanSheepDog, 2019-03-23:
nnn