maximkulkin/esp-homekit-demo #184: IR Homekit repeater

GPL71, 2019-03-18:
@maximkulkin 
Hello ... thank you so much for the great work you doing.
I've a question I'm interested in IR signal repeater homekit compatible.
Did you develop anything about?
Or someone developed something around GitHub?

Thank you again

maximkulkin, 2019-03-18:
Not sure I understand the application of that device in relation to HomeKit. Could you clarify the purpose?

GPL71, 2019-03-18:
@maximkulkin sorry for my english ...
I would like build a HomeKit Accessory compatible to send IR signals to a specific device e.g. TV or A/C or other.
I'm looking for a sample or a tool to grab the raw code of IR and after insert it in a ESP32 and IR transmitter.

I know that someone do it and I imaged it was thank to a your library or sample

I hope now is a little bit clear ... Thank you

peros550, 2019-03-18:
@GPL71 

You could possibly create classic switches in home app that would trigger code which sends  IR commands. 
Have a look on that: 
https://github.com/pbendersky/ir-esp-uart/issues/1

GPL71, 2019-03-18:
@peros550 Thank you so much for your how to .....

maximkulkin, 2019-03-18:
@GPL71

Actually, I AM currently working on an IR library for the same reason. Unlike https://github.com/pbendersky/ir-esp-uart my library uses I2S clock gen and hardware timer interrupts to drive transmission, so it should be more effective. I have transmission part already working, but I'm facing some issues which I think are related to hardware I use. I will receive new hardware (IR diodes) in next few days and will publish library and example then.

peros550, 2019-03-19:
@maximkulkin 

This [article](https://www.analysir.com/blog/2017/01/29/updated-esp8266-nodemcu-backdoor-upwm-hack-for-ir-signals/) provides some interesting information on the hardware design.  

Although you follow a different approach, I believe the circuit could have a lot similarities. If you want help testing the code let us know. I'm sure a few people here (including me) would like to help by testing new features.

maximkulkin, 2019-03-19:
Ok, I've successfully tested initial design and pretty happy with it. Here is a link to project repository:
https://github.com/maximkulkin/esp-ir-thermostat
It uses my own [esp-ir-i2s](https://github.com/maximkulkin/esp-ir-i2s) library to send IR commands. Currently it assumes NEC-like protocol. I plan to add support for raw commands (for those lazy ones who do not want to invest into reverse engineering protocol). Although with my thermostat I had to do that, because it is a stateful communication (it's not just recording and playing button commands).

peros550, 2019-03-19:
Big Thank you! I'm waiting for the RAW support. I tried doing reverse engineering my Toyotomi's protocol in the past but I wasn't successful.

peros550, 2019-03-28:
@maximkulkin ,

I haven't been able to test your library yet (more on this later), but I just had an idea and wanted to let you know. 

I thing the new ios 12.2 brings TV support for Homekit. It would be very nice to have a new example pointing to a TV accessory. With the proper hardware, we could have basic control of TVs using your IR library and home app.

maximkulkin, 2019-04-22:
I guess, everyone is happy now and this can be closed. Let me know if you have more questions

closed 2019-04-22 by maximkulkin