maximkulkin/esp-homekit-demo #185: HomeKit: Failed to read flash magic

AchimPieters, 2019-03-21:
Hey maxim,

Thank you for your good job on this project! I got it almost working. I use life-cycle-manager to upload the bin file to my ESP. Everting works fine until the homekit server want's to start see error below:

```
ESP-Open-SDK ver: 0.0.1 compiled @ Mar 21 2019 19:03:37
phy ver: 273, pp ver: 8.3

mode : sta(60:01:94:2d:3b:0b)
add if0
>>> HomeKit: Starting server
!!! HomeKit: Failed to read flash magic
>>> HomeKit: Formatting flash at 0x100000
!!! HomeKit: Failed to erase flash
>>> HomeKit: Generated new accessory ID: D9:23:BE:37:4C:18
!!! HomeKit: Failed to write accessory ID to flash
>>> HomeKit: Generated new accessory key
!!! HomeKit: Failed to write accessory key to flash
>>> HomeKit: Configuring mDNS
scandone
no mywifi found, reconnect after 1s
reconnect
scandone
no mywifi found, reconnect after 1s
```
Why is this error occurring could you please help me? thanks in advance1

maximkulkin, 2019-03-21:
What kind of device do you have? How much flash memory it has? What is the size of your firmware (you can see the size when you flash it to device). Also, I'm not familiar on memory layout life-cycle-manager uses. @HomeACcessoryKid can you help with that?

AchimPieters, 2019-03-22:
@maximkulkin It's a ESP8266-01 with 1MB flash memory. The size of the bin file is 420 KB.

maximkulkin, 2019-03-22:
Ok, at least I see one problem (still not sure about life-cycle-manager): your HomeKit is configured to store it's data past 1MB mark which is clearly something you do not have. Check step 6 (Configure settings) in this repository's README, you need this:
```
6. Configure settings:

If you have 1MB chip, you need to set following environment variables:

    export FLASH_SIZE=8
    export HOMEKIT_SPI_FLASH_BASE_ADDR=0x7a000
```
Make sure you do a clean build after that (`make clean ; make all`), environment variables are not automatically picked up by build system.

RavenSystem, 2019-03-22:
To combine with LCM, better use:
HOMEKIT_SPI_FLASH_BASE_ADDR = 0x8c000

AchimPieters, 2019-03-22:
@maximkulkin  Thank you for your fast reply! I thought I did this, but are not so sure now. I will try Later when I home from work. 

@RavenSystem  Thank you also for your reply, I will try your suggestion, I see that you already have more experience with LCM. If you have more pointers and /or tips they are welcome!

I will keep you updated!

RavenSystem, 2019-03-22:
@AchimPieters in your log, you can see `Formatting flash at 0x100000`, that means you used `HOMEKIT_SPI_FLASH_BASE_ADDR = 0x100000`.

AchimPieters, 2019-03-23:
I'v tried your suggestions but unfortunately without solving the error. I tried both 0x7a000 and 0x8c000 addresses.

Here is what I did:

open the terminal on my Mac:

`cd esp`
then
`cd esp-homekit-demo`
make clean as suggested:
`docker run -it --rm -v `pwd`:/project -w /project esp-rtos make -C examples/led clean`
hereafter I typed:
`export FLASH_SIZE=8`
and
`export HOMEKIT_SPI_FLASH_BASE_ADDR=0x7a000`
the compiled it again:
`docker run -it --rm -v `pwd`:/project -w /project esp-rtos make -C examples/led all`

pushed my bin files to git for installation. See LCM.

Erased the Falser from my ESP
`esptool.py erase_flash`

Then upload the LCM files. configure Wifi and Git settings. See LCM.

Everything starts and runs perfectly until I get the error:

```
>>> HomeKit: Starting server
!!! HomeKit: Failed to read flash magic
>>> HomeKit: Formatting flash at 0x100000
!!! HomeKit: Failed to erase flash
>>> HomeKit: Generated new accessory ID: F3:2E:1B:39:32:AE
!!! HomeKit: Failed to write accessory ID to flash
>>> HomeKit: Generated new accessory key
!!! HomeKit: Failed to write accessory key to flash
>>> HomeKit: Configuring mDNS
scandone
no mywifi found, reconnect after 1s
```

Am I doing something wrong? :confused:

maximkulkin, 2019-03-23:
Enviroment variables are not automatically exported into Docker container. There are multiple ways to fix that, I prefer just listing settings in `make` command:
```shell
docker run -it --rm -v pwd:/project -w /project esp-rtos make -C examples/led FLASH_SIZE=8 HOMEKIT_SPI_FLASH_BASE_ADDR=0x7a000 all
```

AchimPieters, 2019-03-23:
so it worked, kind off:

```
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x8c000
>>> HomeKit: Generated new accessory ID: 81:27:8A:A8:8D:95
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
scandone
no mywifi found, reconnect after 1s
```

maximkulkin, 2019-03-23:
Please read README, it has a lot of insights.

AchimPieters, 2019-03-23:
Obviously I need to add the credentials in the wifi.h file, when I have don that I get this message and are not able to connect:

```
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 16:91:D5:70:2B:5D
>>> HomeKit: Configuring mDNS
scandone
add 0
aid 2
cnt 
rm match
reconnect
scandone
add 0
aid 2
cnt 
```

In combination with LCM I think I need to clone esp-wifi-config in to my esp directory and then change the led.c file `#include "wifi.h"` -->> `#include <wifi_config.h>`

I think I'am almost there....

AchimPieters, 2019-03-25:
Okay this is how far I have come, adding wifi credentials to the wifi.h file makes that it works, as I thought. But It did't  solve my LCM problem. Then I saw the wifi-config folder. I examined the code, here I saw the wifi_config.h file, I stumble upon it already. Could this be the solution?

I compiled the wifi_config.h code to a bin file and pushed it to git run LCM on my ESP and there it is... the solution.

do I need to change the code for every accessory or is there a simpler way?

AchimPieters, 2019-03-26:
Eureka!  :smiley:  wifi_config.h did the trick!

closed 2019-03-26 by AchimPieters

AlexJanka81, 2019-07-10:
Hi Maxim, I use this topic because it is near my question. I have a MAGIC HOME Module like this: https://www.ryslux.de/images/product_images/popup_images/259_1.jpg

It has an ESP8285 Module on it. I did all steps (RTOS, SDK, ESP homekit demo, changed wifi_config and started MAKE on MAC with an virtual linux machine). It runs so far.
I also initiate a communication to the pre-programmed module. esptool.py detects automatically "8m" flash size. 
Then it downloads the code and it seems to finish the programming, but the behavior of the module does not change. I get no useful prints on minicom and my router does not recognize a new WLAN partner.
I also use 0x7a000 in the esptool.py command.
Later when I am at home (9:00 pm in Germany) I would like to post my commands. Perhaps I did something wrong.

Best regards,
Alex

AlexJanka81, 2019-07-10:
Hi, I tried esptool.py erase_flash... this works. Now it is empty and the behavior of the module changed.... I does not do anything. Of course.
But I do not get the magic home stripe demo running. At which address I have to program the bin file? 

Alex

maximkulkin, 2019-07-10:
Not sure about minicom, esp-open-rtos comes with it's own monitoring software, just do `make monitor` (or `make -C examples/magic_home_strip monitor`, depending on how you work with it). Also, MagicHome firmware is equipped with wifi_config, that means that it won't immediately join your WiFi network, it will create it's own and will wait for you to join it and reconfigure to use your WiFi network. So look for WiFi network like "LED Strip-XXXXXX" (where XXXXXX is some hex numbers). Network configuration is done through a captive portal.

AlexJanka81, 2019-07-10:
Hi, first of all thank you for your reply and thank you for your work.

This is my Output.
user@ubuntu:~/projects/esp-homekit-demo/examples/magic_home_strip/firmware$ 
esptool.py --port /dev/ttyUSB0 --baud 57600 write_flash --flash_mode dout --flash_size 8m 0x7A000 magic_home_strip.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Writing 454656 @ 0x7a000... 454656 (100 %)
Wrote 454656 bytes at 0x7a000 in 78.9 seconds (46.1 kbit/s)...
Leaving...

But it is the same output at 0x7A000 or 0x00000 or 0x01000...

user@ubuntu:~/projects/esp-homekit-demo/examples/magic_home_strip$ make monitor
/home/user/projects/esp-open-rtos//utils/filteroutput.py --port /dev/ttyUSB0 --baud 115200 --elf ./build/magic_home_strip.out
Opening /dev/ttyUSB0 at 115200bps...

and than nothing. CTRL+C breaks it. What should happen?

I also do not find another wifi.
In fact I do not want another wifi. I do not want to use the original SW. I thought I can replace the original firmware with your demo and the module connects to my wifi account, which I hardcoded into the SW with wifi.h and privat_ssid_config.h... 
I also changed the pins in magic_home_strip.c to the pins of the module.
But my linux, Mac and programming knowledge is too small to do improvisation. So all I achieved till now is to erase the flash... :)

Best regards,
Alex

maximkulkin, 2019-07-10:
Can you post your code for inspection?

maximkulkin, 2019-07-10:
Ok, no output in console is because console is disabled, because LED control is done through hardware protocol that is only available on the same pin as serial interface. Debugging those problems is a little more involved. I recommend 1. Enable uart; 2.replace ws2812_i2s library with ws2812 and move data pin away from serial pin; 3. Test everything (you should be able to get console output); 4. Once everything works, undo your changes (switch ws2812 library back, change data pin back, disable uart).

Also, if you want to use static WiFi credentials, you need to replace wifi_config back to manual wifi configuration (you can find example in examples/led).

AlexJanka81, 2019-07-11:
Ok, thank you. I will try it. But it isn't a WS2812 stripe. I use magic_home_strip.c out of the esp-homekit-demo/examples. 
https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/magic_home_strip

It uses multipwm. I just swapped some PWM pins.

Thanks again for reply. I will try it and play around with some settings.

Alex

AlexJanka81, 2019-07-11:
Hi, ok make monitor shows me this:
Fatal exception (28):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28):
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000

always after programming and often till resetting the module.

I also start from the beginning: checked out all needed stuff from git and build led example or http_get example out of the Esp_open_rtos example folder.

the behavior and the printings are the same. 
Alex

maximkulkin, 2019-07-12:
You're doing something wrong. Erase the whole flash, `make clean` and rebuild everything. Make sure to set correct HOMEKIT_SPI_FLASH_BASE_ADDR correclty.

AlexJanka81, 2019-07-12:
Seems so. OK, I will try it. Perhaps the hardware is wrong. GPIO 0 has to pulled low everytime? or just while RESET? I get the module into flashmode, but while flashing GPIO0 is high, cause of a pullup.

AlexJanka81, 2019-07-12:
Hi, LED example out of HomeKit Demo works. I don't no really why, but it is good. I am using "another" make command. It is the one from the rtos instructions: make flash -j4 ESPPORT=/dev/ttyUSB0 and it programs the bin after compiling.

just type "make" and then via command esptool.py .... doesn't work. 

make flash -j4 ESPPORT=/dev/ttyUSB0 also program more things at the same time: 

esptool.py v1.2
esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash -fs 8m -fm dout -ff 40m \
	0x0 /home/user/projects/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /home/user/projects/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/led_aj.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0320
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (88.3 kbit/s)...
Writing 4096 @ 0x1000... 4096 (100 %)
Wrote 4096 bytes at 0x1000 in 0.4 seconds (88.6 kbit/s)...
Writing 430080 @ 0x2000... 430080 (100 %)
Wrote 430080 bytes at 0x2000 in 37.3 seconds (92.2 kbit/s)...
Leaving...

Pairing works. And Homekit works.

Fine. Thank you!!!!

Alex