maximkulkin/esp-homekit-demo #186: Example "Programmable Switch Event" for ESP32

GermanSheepDog, 2019-03-23:
Hello Maxim,

I am looking for an example HOMEKIT_SERVICE_STATELESS_PROGRAMMABLE_SWITCH for an ESP32!
Today there is no example for ESP32 and I started with the template of the esp32/led-example to create an esp32/button-example.

If I started the short example and I look at my iPhone I got the same error as in my big program:
```
  Standardraum Button
  Es gibt eine nicht konfigurierte Taste.

  or in english: 

  Standard room button
  There is an unconfigured button.
```
Here are my esp32-example-directory:
[esp32.zip](https://github.com/maximkulkin/esp-homekit-demo/files/2999461/esp32.zip)

Note: Currently I did not implement the time function to read how the long the button was pressed.

maximkulkin, 2019-04-01:
Have you configured any automation for the button?

GermanSheepDog, 2019-04-05:
I did not configure anything. I use my example and nothing more.
Is there any step missing?

maximkulkin, 2019-04-05:
Well, the button on its own does not do anything, you need to create an automation to do something when you trigger your switch (e.g. https://youtu.be/UyCtxQwTZz0 at 20:00 timecode)

maximkulkin, 2019-04-12:
The accessory block in Home app is not expected to do anything. It’s there just to allow configuring actions. The actions are triggered by pressing physical button attached to a pin.

GermanSheepDog, 2019-04-13:
Hello Maxim,

thank you for the screenshots. My problem started at the third screenshot, there are no actions like single press, double press or long press.

I found the problem in my big program: This part of your button-example..
```
HOMEKIT_SERVICE(
         STATELESS_PROGRAMMABLE_SWITCH,
            .primary=true,
            .characteristics=(homekit_characteristic_t*[]) {
                  HOMEKIT_CHARACTERISTIC(NAME, "Button"),
                  &button_event,
                  NULL
             },
         ...
```
... is very different at my program.

I have to continue analyzing.

maximkulkin, 2019-04-13:
BTW: this screenshots are taken by running your firmware.

FYI I have a HomeKit hub - an always on iOS device that monitors all accessories and runs automations (in my case its Apple TV). Although when I created the button example I didn’t have it but still was able to test it.

GermanSheepDog, 2019-04-14:
BTW: Now I know, that my firmware "esp32.zip" are running. It was my fault.

My original problem was my big program. Now I create a new test program that is closer to my challenge "dynamically generated services" for "switch events" :
```
#include <stdio.h>
#include <string.h>
#include <esp_wifi.h>
#include <esp_event_loop.h>
#include <esp_log.h>
#include <nvs_flash.h>

#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include "wifi.h"
#include "esp32_button.h"

#define BUTTON_PIN 32

#ifndef BUTTON_PIN
   #error BUTTON_PIN is not specified
#endif

homekit_server_config_t config;

void on_wifi_ready();

esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
        case SYSTEM_EVENT_STA_START:
            printf("STA start\n");
            esp_wifi_connect();
            break;
        case SYSTEM_EVENT_STA_GOT_IP:
            printf("WiFI ready\n");
            on_wifi_ready();
            break;
        case SYSTEM_EVENT_STA_DISCONNECTED:
            printf("STA disconnected\n");
            esp_wifi_connect();
            break;
        default:
            break;
    }
    return ESP_OK;
}

static void wifi_init() {
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&wifi_init_config));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));

    wifi_config_t wifi_config = {
        .sta = {
            .ssid = WIFI_SSID,
            .password = WIFI_PASSWORD,
        },
    };

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());
}

[... 102 lines cut ...]
    homekit_server_init(&config);
}

void app_main(void) {
    // Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
       ESP_ERROR_CHECK(nvs_flash_erase());
       ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    wifi_init();
    if (button_create(BUTTON_PIN, button_callback)) {
        printf("Failed to initialize button\n");
    }
}
```
The main part of the test program worked fine:
[image]
But are there are some issues:
- How can I assign "service->characteristics[1] = &button_event;" in "create_new_homekit_service"?
  I must use some if-statements and that is not so good.
- Do you know why my iPhone show "TASTE1" and not the NAME "Button 1"?
Thank you in advance for your help.

GermanSheepDog, 2019-04-18:
Here are a completed example:
[esp32_button.zip](https://github.com/maximkulkin/esp-homekit-demo/files/3094085/esp32_button.zip)

But are there are some issues:

- Do you know why my iPhone show "TASTE 1" and not the NAME "Button 1"?
- The order of the buttons at the iPhone sometime differently, for example: Sometime "TASTE1" is equal the NAME "Button 1", sometimes "TASTE 1" is equal the NAME "Button 3".
- How can I assign and call the service more dynamically?

This part are not so nice:
```
   if (gpio == BUTTON_PIN1) homekit_characteristic_notify(&button_event1, HOMEKIT_UINT8(0));
   if (gpio == BUTTON_PIN2) homekit_characteristic_notify(&button_event2, HOMEKIT_UINT8(0));
   if (gpio == BUTTON_PIN3) homekit_characteristic_notify(&button_event3, HOMEKIT_UINT8(0));
```
This part also:
```
            HOMEKIT_SERVICE(
                STATELESS_PROGRAMMABLE_SWITCH,
                .characteristics=(homekit_characteristic_t*[]) {
                    HOMEKIT_CHARACTERISTIC(NAME, "Button 1"),
                    &button_event1,
                    NULL
                },
            ),
            HOMEKIT_SERVICE(
            ...
                    HOMEKIT_CHARACTERISTIC(NAME, "Button 2"),
                    &button_event2,
            ...       
            HOMEKIT_SERVICE(
            ...     
                 HOMEKIT_CHARACTERISTIC(NAME, "Button 3"),
                    &button_event3,
            ...
```
Thank you in advance for your help.

maximkulkin, 2019-11-05:
You need to store button related info in some sort of structure and pass it as context pointer when you register callbacks (either button callback or characteristic change callback) and inside callbacks you can inspect that structure and figure out what to do.

closed 2019-11-05 by maximkulkin