maximkulkin/esp-homekit-demo #189: Change Non-IO Pins to IO

kunjal83, 2019-03-26:
I have tried checking the library and has also checked the issues but not sure if anyone has assigned the Non-IO Pins to change its function as IO. 
For Example, Tx and Rx Pins. Default function of these pins is to be used as Receiver and Transmitter. However, can we change the function for them as IO pins and use them? 

I am aware them how to do it when writing sketch Arduino but not sure here. 

Thanks.

maximkulkin, 2019-03-26:
I think you should be able to do that the normal way: `gpio_enable(1, GPIO_OUTPUT)` (where `1` is TX pin)

kunjal83, 2019-03-27:
This doesnt work out in case of ESp8266-12F. 
when I write `const int toggle_gpio5 = 5;` and `gpio_enable(toggle_gpio5, GPIO_INPUT);`, this specifically activates GPIO5 on this chip. 

Do you think, we need to additional add anything more here?

maximkulkin, 2019-03-27:
>  this specifically activates GPIO5 on this chip.
What does that mean?!

kunjal83, 2019-03-27:
apologies if thats confusing. What i meant was, based on your comment, `gpio_enable(1, GPIO_OUTPUT)`, writing `const int toggle_gpio5 = 5;` and `gpio_enable(toggle_gpio5, GPIO_INPUT);` should ideally activate GPIO14 but here it activates GPIO5.

maximkulkin, 2019-03-27:
`gpio_enable(5, GPIO_INPUT)` activates GPIO5?! What a surprise!

kunjal83, 2019-03-27:
I am surprised as well and that was the reason to start this thread.

maximkulkin, 2019-03-27:
Try using `gpio_enable(14, GPIO_INPUT)`

kunjal83, 2019-03-27:
Thats activates GPIO14

maximkulkin, 2019-03-27:
Exactly! Here you go!

kunjal83, 2019-03-27:
Yes, but here we go back to my original question, how do I activate the non-GPIO pins for IO related operations?

maximkulkin, 2019-03-27:
I would say: you can not repurpose non-GPIO pins. Here is detailed pinout diagram, you can see that most pins are GPIO pins, except for power/ground ones, chip enable, reset, ADC and two reserved pins.
[image]

kunjal83, 2019-03-28:
No problem. On another note, how can I put the ESP to sleep? 
I think there are 2 sleep modes, Light Sleep and Deep Sleep. Any thoughts?

maximkulkin, 2019-03-28:
No experiments were made so far. As far as I know, deep sleep does not preserve memory state, that means that it's like reboot of device. Also it will be disconnected from network and HomeKit controllers won't be able to communicate with it (will be "Unreachable").

Not sure about Light sleep and Modem sleep. It seems from description, that it should be possible to use Light sleep without loosing connectivity.

kunjal83, 2019-03-29:
Let me experiment this weekend and feedback.

kunjal83, 2019-03-29:
@maximkulkin, can you quickly confirm if the framework has setup for sleep already done? if yes, can you point me to the library please?

maximkulkin, 2019-03-29:
Not sure what kind of setup you expect.

kunjal83, 2019-03-29:
i am talking about the below 2 command. I am unable to find anything in the current library related to these commands.

`wifi_fpm_open();
 wifi_fpm_do_sleep(26843455);`

maximkulkin, 2019-03-29:
IMO this is not something HomeKit framework should care about. Moreover, those functions do not exist in esp-open-rtos.

kunjal83, 2019-03-29:
No problem. I will try and see what can be done.

maximkulkin, 2019-04-04:
Seems like original issue was resolved, closing this issue

closed 2019-04-04 by maximkulkin