maximkulkin/esp-homekit-demo #192: Question: Weather Station Sensors

zacol, 2019-03-28:
I was thinking about building weather station based on ESP-32. I've got a couple of sensors: 
- temperature and humidity sensor DHT-22,
- rain sensor YL-80, 
- soil moisture sensor,
- fotorezistor sensor. 

DHT-22 shouldn't be a problem at all.

I know that Homekit doesn't support rain sensors but maybe there is an option to expose them somehow. Maybe as switch with two states - rain and not rain... I'm guessing. 

I don't know what about soil moisture and fotorezistor sensors. Are they supported by Homekit?

maximkulkin, 2019-03-28:
I guess those sensors are not present in HomeKit, because they do not make much sense for a smart home to implement automation tasks. Your accessory can still you them internally. If you do want to expose those in HomeKit and build automations with them, you can try to match to existing HomeKit sensors. E.g. binary sensors like fotorezistor sensor can be implemented as stateless programmable switch.

closed 2019-04-01 by maximkulkin

zacol, 2019-04-13:
@maximkulkin I thought that you may be interested in my progress. I've started a new repository and at the moment I was able to connect soil moisture sensor, light sensor, temperature and humidity sensor.

It's the first time when I'm writing in C so any insights will be appreciated.
https://github.com/zacol/glasshouse

maximkulkin, 2019-04-22:
Looks good!