maximkulkin/esp-homekit-demo #193: Esp8266 512k

beckmx, 2019-04-04:
Hello, i have some esp01 laying around and would like to know if they can support a compilation made with the entire toolchain, for instance the led example and what could be the possible flash values

Awesome work :)

maximkulkin, 2019-04-04:
Wow, that’s a rare piece, most esp-01 I can find have at 1MB of flash. Although you should be fine if you set SPI base address to 0x7a000 (see instructions in README)

beckmx, 2019-04-07:
at first didnt work with the readme but then I noticed I would also need to decrease the other configuration:
export FLASH_SIZE=4
export HOMEKIT_SPI_FLASH_BASE_ADDR=0x7a000

With these configs, looks everything is working nicely =D

beckmx, 2019-04-07:
Ok, I noticed something weird, it does create device id and passes the pairing process but after rebooting it fails and stucks, I guess is because of memory refs?

maximkulkin, 2019-04-07:
Logs please

beckmx, 2019-04-08:
I cannot get logs from device it just shows undefined characters “□”. This is how I was testing:
1) Compile exactly as described, with these values:
export FLASH_SIZE=4
export HOMEKIT_SPI_FLASH_BASE_ADDR=0x7a000
my esptool params are:
esptool.py \
            -p /dev/port \
            --baud 115200 \
            write_flash \
            --flash_size 512KB \
            --flash_mode qio \
            --flash_freq 40m \
            0x0 ./firmware/rboot.bin \
            0x1000 ./firmware/blank_config.bin \
            0x2000 ./firmware/wifi_config.bin
If I do this, I will get messages in the console like "device id not generated" and also not able to save the keys from the pairing process

2) if I do the same as step1 but --flash_size 1MB it does generate the device id and also able to save keys from the pairing process BUT if I reboot it I will only get some characters “□” and dies

Do you know more or less what is the flash map, maybe it needs more than 512kb

wifijt, 2019-04-27:
I was running into these issues as well 
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
!!! HomeKit: Failed to compact data: sector data read error
!!! HomeKit: Failed to write pairing info to flash: max number of pairings
!!! HomeKit: [Client 4] Failed to store pairing (code -2)
>>> HomeKit: [Client 4] Closing client connection

I was able to make this work with the following settings 

esptool.py -p /dev/tty.usbserial-1451120 --baud 115200 write_flash -fs 2MB -fm dout -ff 40m

AchimPieters, 2019-05-03:
I'll hope you get this working! because maybe it's possible to change the module to a YJ-16013 BLE module. This would be awesome with the [Temperature sensor](https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/temperature_sensor). I would be able tu run on a battery!

YJ-16013 BLE module is based on Nordic nRF52832 Soc , The nRF52832 SoC is a powerful, highly flexible ultra-low power multiprotocol SoC ideally suited for Bluetooth® low energy (previously called Bluetooth Smart), ANT and 2.4GHz ultra low-power wireless applications. The nRF52832 SoC is built around a 32-bit ARM® Cortex™-M4F CPU with **512kB** + 64kB RAM. The embedded 2.4GHz transceiver supports Bluetooth low energy, ANT and proprietary 2.4 GHz protocol stack. It is on air compatible with the nRF51 Series, nRF24L and nRF24AP Series products from Nordic Semiconductor.

closed 2019-05-13 by maximkulkin