maximkulkin/esp-homekit-demo #194: info magichome led strip

curla92, 2019-04-04:
Hi maxim. I would to know which led strip you used to test your magichome firmware. I tried with a simple rgb led strip but it doesn't work well. Colors are wrong and i was not able to resolve the problem. Thanks and i really appreciate your work.

maximkulkin, 2019-04-04:
Colors might be wrong because different RGB LEDs have different color orders (e.g. RGB, GRB, GBR).. You just need to figure which one is yours and adjust. You can create a simple firmware that first sets all RED, then GREEN, then BLUE and notice the actual colors.

curla92, 2019-04-04:
What's the GPIO scheme?

maximkulkin, 2019-04-04:
https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/led_strip/led_strip.c#L2-L9

curla92, 2019-04-05:
i meant gpio scheme of magic home thank y

maximkulkin, 2019-04-05:
What do you call gpio scheme?!

curla92, 2019-04-07:
with espurna firmware my led strip works with this configuration.
    #define LIGHT_CH1_PIN       12      // RED
    #define LIGHT_CH2_PIN       5       // GREEN
    #define LIGHT_CH3_PIN       13      // BLUE
    #define LIGHT_CH4_PIN       14      // WHITE

closed 2019-04-22 by maximkulkin