maximkulkin/esp-homekit-demo #195: Somethings missing in esp-homekit-demo/components/common/homekit ?

jonasborneland, 2019-04-07:
Im getting this error
```
~/projects/esp-homekit-demo$ make -C examples/led test
make: Entering directory '/home/user/projects/esp-homekit-demo/examples/led'
/home/user/projects/esp-open-rtos/common.mk:223: /home/user/projects/esp-homekit-demo/components/common/homekit/component.mk: No such file or directory
```

And also this:
```
~/projects/esp-homekit-demo$ git submodule update --init --recursive
fatal: reference is not a tree: 44d5dda87e0991701e9c0897b5afe55b659183cf
Unable to checkout '44d5dda87e0991701e9c0897b5afe55b659183cf' in submodule path 'components/common/homekit'
```

Seems like there is nothing in the folder "esp-homekit-demo/components/common/homekit/"
The folder here leads to a 404 page https://github.com/maximkulkin/esp-homekit-demo/tree/master/components/common

maximkulkin, 2019-04-08:
You probably did not follow instructions in README and didn’t do `git submodule update --init --recursive`

jonasborneland, 2019-04-08:
@maximkulkin Thanks for the answer and for the awesome repository! :)
I did use `git submodule update --init --recursive` look at the post above.

I did also make it work using the `git checkout` command and choose an old version.

I think there is missing something in your online version of the update (the one you uploaded 3 days ago). Please take a look at the "common/homekit/" ... There is no files in the folder online

maximkulkin, 2019-04-08:
Oh, indeed, I forgot to push latest commits to esp-homekit. Thanks for pointing that out. I updated both esp-homekit and this repository.

closed 2019-04-08 by maximkulkin

jonasborneland, 2019-04-08:
@maximkulkin Awesome! Thanks! Keep up the good work 😃