maximkulkin/esp-homekit-demo #196: Could not initialize sysparams

duhmojo, 2019-04-08:
Hi. I'm really keen to try the garage example out. On an Ubuntu VM I was able to successfully setup and build the example garage firmware, but it doesn't appear to want to initialize. When I make "test" the result is "Could not initialize sysparams (1)". Any tips? I'm using a Adafruit Feather with the ESP8266 chip. I noticed that there's a wemos.h which I figured I would need to reassign for the feather.

Also, is a home kit .setup pin needed in homekit_server_config_t?

$ make -C examples/garage test
make: Entering directory '/home/chris/Development/esp-homekit-demo/examples/garage'
CC /home/chris/Development/esp-open-rtos/open_esplibs/libmain/timers.c
CC /home/chris/Development/esp-open-rtos/open_esplibs/libmain/ets_timer.c
CC /home/chris/Development/esp-open-rtos/open_esplibs/libmain/spi_flash.c
...
LD build/garage.out
FW firmware/garage.bin
esptool.py v1.2
esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash -fs 32m -fm qio -ff 40m \
	0x0 /home/chris/Development/esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin 0x1000 /home/chris/Development/esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/garage.bin 
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.9 kbit/s)...
Writing 4096 @ 0x1000... 4096 (100 %)
Wrote 4096 bytes at 0x1000 in 0.4 seconds (90.0 kbit/s)...
Writing 434176 @ 0x2000... 434176 (100 %)
Wrote 434176 bytes at 0x2000 in 37.6 seconds (92.3 kbit/s)...
Leaving...
/home/chris/Development/esp-open-rtos//utils/filteroutput.py --port /dev/ttyUSB0 --baud 115200 --elf ./build/garage.out
Opening /dev/ttyUSB0 at 115200bps...
Booting rom 0.
WA�%9� Could not initialize sysparams (1)!

Also, on Ubuntu I had to enable more permissive permissions on ttyUSB to use the make file flash/test options:

$ sudoedit /etc/udev/rules.d/50-myusb.rules
KERNEL=="ttyUSB[0-9]*",MODE="0666"
KERNEL=="ttyACM[0-9]*",MODE="0666"

You might want to mention something about esptool usually requiring root/sudo.

Thanks for the awesome work!

maximkulkin, 2019-04-08:
Try doing `make erase_flash` first.

duhmojo, 2019-04-08:
I’ll give it a try today. Thanks for the quick reply.

The 2nd part isn’t a question, it’s a documentation suggestion. If,when you try running esptool manually you find you have to sudo, then you’ll need to be more permissive with access to ttyUSB. It isn’t special to a VM (USB is passed through) and the only thing Ubuntu specific is maybe the default direct ttyUSB permissions.

duhmojo, 2019-04-08:
Using "erase_flash" did the trick. I was able to find and add the Garagentor accessory, but I had to "flash" then "monitor" separately. If I "test" the serial output I get dead air. No problem, just letting you know what I experienced. So good progress!

I'm playing with a Adafruit Feather. How does its pins compare to the Wemos D1 pins in the wemos.h? Is there any hardware wiring for the examples provided? I mean there's little word about the Wemos D1 anywhere except the in source, but there are comments about 10k resistors.

I was hoping to add a reed magnetic switch to detect if the garage door was open, and a relay to engage the wiring to the existing wall button.

closed 2019-04-11 by duhmojo

MizzyAntunes, 2020-03-03:
> Try doing `make erase_flash` first.
> 
> For the second question: I'm not aware about inner workings of Ubuntu inside VM and USB support.

I tried to use this solution and received the following message:

vagrant @ vagrant-ubuntu-trusty-64: ~ $ make erase_flash
make: *** No rule to make target `erase_flash '. Stop.

MizzyAntunes, 2020-03-03:
And I found this after "vagrant up"

==> default: Machine already provisioned. Run `vagrant provision` or use the `--provision`
==> default: flag to force provisioning. Provisioners marked to run always will still run.

MizzyAntunes, 2020-03-03:
Another error

vagrant@vagrant-ubuntu-trusty-64:~/esp-homekit-demo$ make -C examples/led test
make: Entering directory `/home/vagrant/esp-homekit-demo/examples/led'
CC /home/vagrant/esp-open-rtos/open_esplibs/libmain/ets_timer.c
make: xtensa-lx106-elf-gcc: Command not found
make: *** [build/open_esplibs_libmain//ets_timer.o] Error 127
make: Leaving directory `/home/vagrant/esp-homekit-demo/examples/led'