maximkulkin/esp-homekit-demo #199: Garage example reversed reed switch states?

duhmojo, 2019-04-11:
Hi. The REED_PIN state seems to be reversed in current_door_state_update_from_sensor(). Here's log output when the switch was "closed", then when it was "open". The contact sensor state is correct but current_door_state_update_from_sensor() reverses it in the sensor_state switch statement (garage.c line 195).

```
contact sensor state 'closed'.
returning current door state 'open'.
Returning target door state 'open'.
Notifying homekit that target door state is now 'open'
Notifying changed 'Target Door State'
Notifying homekit that current door state is now 'open'
Notifying changed 'Current Door State'

contact sensor state 'open'.
returning current door state 'closed'.
Returning target door state 'closed'.
Notifying homekit that target door state is now 'closed'
Notifying changed 'Target Door State'
Notifying homekit that current door state is now 'closed'
Notifying changed 'Current Door State'
```

Shouldn't the state returned to Homekit not match the reed switch? It's not just an observation,  the reed switch would be open when the garage door is up and the switch is disconnected. Maybe you were using different type of reed switch/sensor?

maximkulkin, 2019-04-11:
I guess it depends on where you put your reed switch: you can put it at the bottom of garage door so that when door is down (closed), the switch will also be closed. On the other hand, you can put your switch on top of the garage door so that when it's down (closed), the switch will be open (untriggered). Then when door opens, it triggers the switch and it becomes closed while the door is opened.

You can add a compiler define to alternate the logic to suit your needs (or just edit it on the spot).

closed 2019-04-11 by maximkulkin

duhmojo, 2019-04-11:
Thanks for confirming maximkulkin. I wasn’t sure yet if this was expected and I somehow messed up my reed switch wiring. Thanks for the great work.

dachoeks3, 2021-05-12:
Not really following this... How do you reverse logic?